GPUOpen-Drivers/AMDVLK #294: Detroit : Become Human render artifacts (Lens Flare)

alou-S, 2022-09-15:
In the game Detroit: Become Human when using either the `AMDVLK` or the `AMDGPU-PRO` drivers causes the lens flares in the game to be rendered black. This causes major visual problems since it almost always covers a huge part of the screen in black. The problem doesn't seem to happen using the `RADV` driver but said driver performs horribly on this game.

Example Video of said artifac

https://user-images.githubusercontent.com/73123610/190375608-1bf8a6f7-1154-4565-8c5f-a1fdb9cc1005.mp4

ts:

jinjianrong, 2022-09-19:
@alou-S we can't reproduce the issue. what GPU are you using?

alou-S, 2022-09-19:
RX 6600M

I have tried reinstalling the entire distro but even that doesn't help. IDK the chances the bug can be related to Arch but I could try using another distro.

jinjianrong, 2022-09-19:
We finally reproduced the issue. Will check it.

alou-S, 2022-10-12:
@jinjianrong Just bumping to keep it fresh

jinjianrong, 2022-10-13:
Thanks for the reminder. we are working on it.

alou-S, 2022-12-18:
@jinjianrong Any updates on why is it taking so long? :/

jinjianrong, 2022-12-19:
@alou-S  Per our developer's investigation, we suspect it might be app's issue. A vertex shader module's input variable is initialized as a Nan value. we could also reproduce the issue with RADV with the renderdoc trace captured from the game.

alou-S, 2022-12-19:
Hmm. Another thing is that when I originally tried this game I had no such lens flare issues when using the AMDVLK Drivers. It started a couple of days later. And after the issue started nothing would fix it.
Reinstalling the Game or Linux.

Also the issue doesn't seem to happen on Windows

alou-S, 2023-01-07:
@jinjianrong So while I was debugging the games low performance issues on the `RADV` drivers I came across some old issues and commits that fix issues with the game. Mainly rendering issues and pixel shading issues.  

One of the simpler patches was to enable `radv_zero_vram` in the `driconf`. I disabled it and was able to perfectly reproduce the exact same bug on `RADV` Drivers without a RenderDoc.  

I guess having a similar implementation to `radv_zero_vram` enabled in `AMDVLK` should fix the issue?

alou-S, 2023-01-07:
Okay looks like after a bit of poking around in the `AMDVLK` drivers and referencing the `RADV` drivers I found the required setting to enable in `AMDVLK` driver.  

The `ClearAllocatedLfb` in AMDVLK directly corresponds to `zerovram` in `RADV`. Enabling it seems to fix the render artifacts.

jinjianrong, 2024-03-25:
workarounded in driver by turning on clear-allocated-local-fb

closed 2024-03-25 by jinjianrong