GPUOpen-Drivers/AMDVLK #298: GPU hang with the vkd3d-proton raytracing tests

mbriar, 2022-09-30:
Running the [vkd3d-proton](https://github.com/HansKristian-Work/vkd3d-proton) raytracing tests hangs the GPU on amdvlk.  
To reproduce build and run the vkd3d-proton tests:  
```
git clone --recurse-submodules https://github.com/HansKristian-Work/vkd3d-proton
mkdir vkd3d-proton/build && cd vkd3d-proton/build
meson .. -Denable_tests=true
ninja
VKD3D_CONFIG=dxr11 VKD3D_TEST_FILTER=raytracing ./tests/d3d12
```
  
GPU 6700XT/NAVI22
AMDVLK v-2022.Q3.5
vkd3d-proton commit 212ae62b5

Flakebi, 2022-09-30:
There’s an internal issue about this since a while.
This vkd3d-proton issue has some information about the problem: https://github.com/HansKristian-Work/vkd3d-proton/issues/817

mbriar, 2022-09-30:
Oh, i wasn't aware of that, feel free to close this issue then. But it seems like it hasn't moved forward on either the vkd3d-proton or the AMD driver side.

Flakebi, 2022-09-30:
I think it’s an amdvlk bug, so this should stay open until the issue is fixed. Thanks for testing and bringing this up!

jinjianrong, 2022-10-13:
@mbriar what Linux distro are you using?

mbriar, 2022-10-13:
This was on Arch Linux with either the [package](https://archlinux.org/packages/extra/x86_64/amdvlk/) or a self-compiled build.

jinjianrong, 2024-10-31:
Fixed with 2024.Q4.1

closed 2024-10-31 by jinjianrong