GPUOpen-Drivers/AMDVLK #304: Feature request: Fine tuning for Cyberpunk 2077

Zeioth, 2022-11-14:
The game runs fine with raytracing, which is incredible. But it seems to compile shaders way too often. IDK if there is some kind of sheader cache size limit on this driver, or it's just performance stuttering.

I performed my tests with the setup:

* Steam Proton
* Ryzen 7950X
* 6700XT

In game settings:

* Native resolution 4K
* Game resolution 1080p
* FSR 2.1 enabled

I easily get 70-90FPS pretty smooth, but ocasionally the game frozes for way too long. 5-10s. And then it keeps going smooth. But it happens often.

zaps166, 2022-11-15:
I have very low performance in Cyberpunk 2077 with AMDVLK (2022.Q4.2). RX 6900XT and i7-8700k, 3840x2160, FSR 2.1 set to image quality:
- AMDVLK - 11 FPS (very high CPU usage)
- RADV - 54 FPS (low CPU usage) - similar performance on Windows

jinjianrong, 2024-03-25:
This should already be fixed in new release

closed 2024-03-25 by jinjianrong

S16201512, 2024-08-26:
> 新版本中应该已经修复了这个问题

which vision? please add the link

qiaojbao, 2024-08-26:
> > 新版本中应该已经修复了这个问题
> 
> which vision? please add the link

https://github.com/GPUOpen-Drivers/AMDVLK/releases
Try the latest one, please.