GPUOpen-Drivers/AMDVLK #310: Cannot create Vulkan instance on RX 7900 XTX

Aldrog, 2022-12-21:
`vulkaninfo` output:
```
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
ERROR: [Loader Message] Code 0 : terminator_CreateInstance: Found no drivers!
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
ERROR at /build/vulkan-tools/src/Vulkan-Tools-1.3.226/vulkaninfo/vulkaninfo.h:672:vkCreateInstance failed with ERROR_INCOMPATIBLE_DRIVER
```

My setup:
OS: Arch Linux (tested with kernels 6.0.12.arch1-1 and 6.1.arch1-1)
CPU: AMD Ryzen 7 7700X (with integrated graphics on `auto` but monitors plugged to discrete GPU only)
GPU: XFX Merc 310 Black (Radeon RX 7900 XTX)
AMDVLK: v2022.Q4.4

`AMD_VULKAN_ICD=RADV vulkaninfo --summary` works properly and displays both integrated and discrete GPUs.

Aldrog, 2022-12-21:
Tested with integrated graphics disabled and now AMDVLK works properly, thus changing the title.

ShobuPrime, 2022-12-27:
I'm a similar issue on on my machine as well. 7950X iGPU:

```
vulkaninfo
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
[1]    8119 segmentation fault (core dumped)  vulkaninfo
```

`AMD_VULKAN_ICD=RADV vulkaninfo --summary` works:

```
Devices:
========
GPU0:
apiVersion         = 4206822 (1.3.230)
driverVersion      = 92286977 (0x5803001)
vendorID           = 0x1002
deviceID           = 0x164e
deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
deviceName         = AMD Radeon Graphics (RADV GFX1036)
driverID           = DRIVER_ID_MESA_RADV
driverName         = radv
driverInfo         = Mesa 22.3.1
conformanceVersion = 1.3.0.0
deviceUUID         = <redacted>
driverUUID         = <redacted>
```

jokeyrhyme, 2023-01-16:
Sounds like this might be the same as the root cause in #305

jinjianrong, 2023-01-17:
The issue might be the 7950X iGPU support has not been exposed in amdvlk. We will get this done in next release

jokeyrhyme, 2023-05-03:
I can confirm that this issue is fixed for me after upgrading to 2023.Q2.1 https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2023.Q2.1

Although, this was likely fixed in 2023.Q1.1 (version was not released for my distribution, so not tested by me, personally)

closed 2023-05-04 by jinjianrong

MasterJH5574, 2023-07-30:
Hi, I’m also encountering the issue. More specifically, `vulkaninfo --summary` fails with the same error msg, while `sudo vulkaninfo --summary` works perfectly well. I’m now remote to my machine and thus is unable to check if disabling the integrated GPU can help. On the other hand, the phenomenon that “only sudo works” confuses me quite a lot. 

My machine setup:
* OS: Ubuntu 22.04
* CPU: AMD Ryzen 7950
* GPU: 2 x AMD Radeon RX 7900 XTX
* AMDVLK: 2023.Q2.3

---
`vulkaninfo --summary` without sudo shows:
```
WARNING: [Loader Message] Code 0 : terminator_CreateInstance: Failed to CreateInstance in ICD 0.  Skipping ICD.
[1]    69432 segmentation fault (core dumped)  AMD_VULKAN_ICD=AMDVLK vulkaninfo --summary
```

`sudo vulkaninfo --summary` shows:
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.250

Instance Extensions: count = 21
-------------------------------
VK_EXT_acquire_drm_display             : extension revision 1
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_portability_enumeration         : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_surface_protected_capabilities  : extension revision 1
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6
VK_LUNARG_direct_driver_loading        : extension revision 1

Instance Layers: count = 12
---------------------------
VK_LAYER_AMD_switchable_graphics_64 AMD switchable graphics layer                       1.3.252  version 1
VK_LAYER_INTEL_nullhw               INTEL NULL HW                                       1.1.73   version 1
VK_LAYER_KHRONOS_profiles           Khronos Profiles layer                              1.3.250  version 1
VK_LAYER_KHRONOS_shader_object      Shader object layer                                 1.3.250  version 1
VK_LAYER_KHRONOS_synchronization2   Khronos Synchronization2 layer                      1.3.250  version 1
VK_LAYER_KHRONOS_validation         Khronos Validation Layer                            1.3.250  version 1
VK_LAYER_LUNARG_api_dump            LunarG API dump layer                               1.3.250  version 2
VK_LAYER_LUNARG_gfxreconstruct      GFXReconstruct Capture Layer Version 0.9.20-unknown 1.3.250  version 36884
VK_LAYER_LUNARG_monitor             Execution Monitoring Layer                          1.3.250  version 1
VK_LAYER_LUNARG_screenshot          LunarG image capture layer                          1.3.250  version 1
VK_LAYER_MESA_device_select         Linux device selection layer                        1.3.211  version 1
VK_LAYER_MESA_overlay               Mesa Overlay layer                                  1.3.211  version 1

Devices:
========
GPU0:
	apiVersion         = 1.3.252
	driverVersion      = 2.0.271
	vendorID           = 0x1002
	deviceID           = 0x164e
	deviceType         = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
	deviceName         = AMD Radeon Graphics
	driverID           = DRIVER_ID_AMD_OPEN_SOURCE
	driverName         = AMD open-source driver
	driverInfo         = 2023.Q2.3 (LLPC)
	conformanceVersion = 1.3.3.1
	deviceUUID         = 00000000-1600-0000-0000-000000000000
	driverUUID         = 414d442d-4c49-4e55-582d-445256000000
GPU1:
	apiVersion         = 1.3.252
	driverVersion      = 2.0.271
	vendorID           = 0x1002
	deviceID           = 0x744c
	deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName         = Radeon RX 7900 XTX
	driverID           = DRIVER_ID_AMD_OPEN_SOURCE
	driverName         = AMD open-source driver
	driverInfo         = 2023.Q2.3 (LLPC)
	conformanceVersion = 1.3.3.1
	deviceUUID         = 00000000-0300-0000-0000-000000000000
	driverUUID         = 414d442d-4c49-4e55-582d-445256000000
GPU2:
	apiVersion         = 1.3.252
	driverVersion      = 2.0.271
	vendorID           = 0x1002
	deviceID           = 0x744c
	deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
	deviceName         = Radeon RX 7900 XTX
	driverID           = DRIVER_ID_AMD_OPEN_SOURCE
	driverName         = AMD open-source driver
	driverInfo         = 2023.Q2.3 (LLPC)
	conformanceVersion = 1.3.3.1
	deviceUUID         = 00000000-0600-0000-0000-000000000000
	driverUUID         = 414d442d-4c49-4e55-582d-445256000000
```

---

Using `AMD_VULKAN_ICD=RADV vulkaninfo` is good, but I want to use AMDVLK so using RADV doesn't address my issue.
Not sure if I can get some help here. Thanks in advance!

MasterJH5574, 2023-07-30:
I have figured it out. Adding the user to both group `video` and group `render` can address the issue and make `vulkaninfo` output properly, though I don't understand the reason.