GPUOpen-Drivers/AMDVLK #313: Request: support for `R8G8B8` vertex attribute formats

glebov-andrey, 2023-02-02:
Currently, both the proprietary driver and RADV support them (except sRGB), but the open source driver does not.

jinjianrong, 2023-02-13:
Per our dump of vulkaninfo using opensource drivers, R8G8B8 vertex attribute formats (except sRGB) are supported, including:

                FORMAT_R8G8B8A8_USCALED
                FORMAT_R8G8B8A8_SSCALED
                FORMAT_R8G8B8A8_UNORM
                FORMAT_R8G8B8A8_SNORM
                FORMAT_R8G8B8A8_UINT
                FORMAT_R8G8B8A8_SINT

@glebov-andrey, how do you find the formats are not supported?

glebov-andrey, 2023-02-13:
@jinjianrong Those are 4-component RGBA formats. I was referring to 3-component RGB formats, specifically:
```
R8G8B8_SNORM
R8G8B8_SSCALED
R8G8B8_UINT
R8G8B8_UNORM
R8G8B8_USCALED
```
Here's a GPUinfo report for AMDVLK on an RX 6900 XT, which does not support any `R8G8B8` formats as vertex attributes: https://vulkan.gpuinfo.org/displayreport.php?id=17900#formats_buffer.
For reference, here's a report for the proprietary driver: https://vulkan.gpuinfo.org/displayreport.php?id=17902#formats_buffer.

jinjianrong, 2023-02-27:
A change to add R8G8B8 support in llpc is under review: https://github.com/GPUOpen-Drivers/llpc/pull/2315

glebov-andrey, 2023-07-31:
Thanks, it works now in 2023.Q2.3.

closed 2023-07-31 by glebov-andrey