GPUOpen-Drivers/AMDVLK #315: ERROR_OUT_OF_HOST_MEMORY when there isn't a suitable AMD GPU in the host system

RyuzakiKK, 2023-02-13:
## Description

If you install AMDVLK in a system where there isn't a suitable AMD GPU, Vulkan programs might crash when they try to load the library `amdvlk64.so`.

## Steps to reproduce

- Use a PC with an Intel or NVIDIA GPU
- Install your expected GPU Vulkan drivers (e.g. on Arch Linux `vulkan-intel`)
- Launch `vulkaninfo`
- Install `amdvlk` (I'm using Arch Linux with `amdvlk 2022.Q4.4`)
- Launch again `vulkaninfo`

## Expected behavior

`vulkaninfo` should always work.

## Actual behavior

After you install AMDVLK, `vulkaninfo` will fail with `vkCreateInstance failed with ERROR_OUT_OF_HOST_MEMORY`.

Full log of `VK_LOADER_DEBUG=all vulkaninfo` https://gist.github.com/RyuzakiKK/5cf7532006491aa2b993fee0e36e548d

## Additional info

This seems to be the same problem that has been reported a few years ago in https://github.com/GPUOpen-Drivers/AMDVLK/issues/187. Maybe the fix regressed in a recent AMDVLK update?

WenqingLiAMD, 2023-03-23:
could you try 2023.Q1.3 please?

jinjianrong, 2024-03-26:
Issue is not observed with v-2024.Q1.2 any more

closed 2024-03-26 by jinjianrong