GPUOpen-Drivers/AMDVLK #317: Missing VK_EXT_image_drm_format_modifier

nyanmisaka, 2023-03-18:
`vulkan: physical device does not support DRM format modifiers`

`VK_EXT_image_drm_format_modifier` is required for importing image from dma buf that contains modifiers.

Since these two related extensions have been supported in AMDVLK, it'd be great if we can have modifier support.
- VK_EXT_external_memory_dma_buf 
- VK_KHR_external_memory_fd

Thanks!

jinjianrong, 2023-03-20:
We have been working on the extension.

nyanmisaka, 2023-03-20:
@jinjianrong Nice to hear that! Any plans on adding modifier in KMD for gfx8 Polaris? This generation still has a wide user base, but due to the big difference from gfx9, developers from Google only added modifier support for gfx9+.

- https://patchwork.freedesktop.org/patch/247020/
- https://patchwork.freedesktop.org/series/80262/
- https://gitlab.freedesktop.org/mesa/mesa/-/issues/5882

nyanmisaka, 2023-08-07:
This can be closed by https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2023.Q3.1

closed 2023-08-07 by nyanmisaka