GPUOpen-Drivers/AMDVLK #320: VKD3D The Last of Us Part 1 D3D_FEATURE_LEVEL_12_0

Hadrianneue, 2023-04-01:
game doesn't seem to detected DirectX 12 while using AMDVLK+AMDGPU, issue not present on RADV, even using VKD3D_FEATURE_LEVEL=12_0 (or 1,2), error message:

[image]

would be great if it would work with TLOU, especially because this title is so vram hungry and RADV is terrible when VRAM is exhausted (really helps in Returnal)

not sure if i should report it here or on VKD3D but other DX12 games runs just fine with AMDVLK+VKD3D like unreal engine games (returnal) or horizon zero dawn (crashes a bunch though), etc.

AMDVLKQ4.4-3/AMDGPU/Mesa-git
EndeavourOS - linux 6.2.9
RX 6500 XT

Hadrianneue, 2023-04-01:
fixed on VKD3D, shader model support 6.6 added

closed 2023-04-01 by Hadrianneue

mbriar, 2023-04-03:
This is not fixed, amdvlk is still lacking some features for shader model 6.6 on vkd3d-proton without faking it with overrides, but there is already a bugreport for it: https://github.com/GPUOpen-Drivers/AMDVLK/issues/289

Hadrianneue, 2023-04-03:
> This is not fixed, amdvlk is still lacking some features for shader model 6.6 on vkd3d-proton without faking it with overrides, but there is already a bugreport for it: #289

it launches though, but thanks for correcting it.

ruiminzhao, 2023-04-06:
@mbriar From the description, I haven't seen any info that can specify this issue is related with #289.  Maybe you mean #289 has blocked "the shader model 6.6 on vkd3d-proton"?  BTW, #289 This issue has already been fixed and delivered, you can check on the commit-id bf720037b81b648e20c54fe1c499b30c11964ef4(under stg/xgl), to enable this extension, you need to add `VKI_COMPUTE_SHADER_DERIVATIVES` to build the related XGL driver(amdvlk64.dll).

mbriar, 2023-04-06:
@ruiminzhao yes, vkd3d-proton requires VK_NV_compute_shader_derivatives to expose D3D12 shader model 6.6 to applications and The Last of Us Part I fails to start if shader model 6.6 is not exposed. VK_NV_compute_shader_derivatives is not yet available in latest AMDVLK 2023.Q1.3 release, but I'm glad to know that it is probably coming soon.

samlupeng, 2023-04-06:
@Hadrianneue 
fixed on VKD3D, shader model support 6.6 added  <-- which version of VKD3D fixed this issue? thanks!

mbriar, 2023-04-06:
@samlupeng there is no vkd3d-proton release that includes this yet, and the fix was only to add an environment variable to override the reported shader model https://github.com/HansKristian-Work/vkd3d-proton/commit/eae3498030da0a16f9a65cd02dbede359e4e6e86

So on vulkan driver that don't support all the necessary extensions for shader model 6.6 it is still required to set VKD3D_SHADER_MODEL=6_6 to launch The Last of Us. See here for which features vkd3d-proton requires to expose shader model 6.6 without any overrides: https://github.com/HansKristian-Work/vkd3d-proton/blob/eae3498030da0a16f9a65cd02dbede359e4e6e86/libs/vkd3d/device.c#L6724