GPUOpen-Drivers/AMDVLK #322: Driver crash on vkCreateRayTracingPipelinesKHR when stageCount = 0 or groupCount = 0

Neo-Zhixing, 2023-04-21:
`vkCreateRayTracingPipelinesKHR` seems to incorrectly assume that `stageCount > 0` and `groupCount > 0`, which are actually allowed by Vulkan specs. One might want to create a ray tracing pipeline entirely from pipeline libraries. However, calling `vkCreateRayTracingPipelinesKHR` like this creates a segfault.

Adding a dummy shader stage and general shader group to `VkRayTracingPipelineCreateInfoKHR` resolves the issue.

This can be reproduced both on Linux and on Windows.

On Windows, this produces a stack frame that looks like this: 
[image]

On Ubuntu, this causes the screen to become completely frozen.

The issue was reproduced on all AMD GPUs I own: RX6600XT, RX6700S, and the integrated graphics in Ryzen 9 6900S.

jinjianrong, 2023-05-18:
@Neo-Zhixing could the issue be reproduced on Linux with amdvlk opensource driver? we find an issue in driver and have a fix, but it is in our proprietary shader compiler and only impacts our closed source driver. do you have a simple app for us to verify the fix?

jinjianrong, 2023-07-27:
Please check the fix with latest release

closed 2023-07-27 by jinjianrong