GPUOpen-Drivers/AMDVLK #324: VKD3D-Proton Support Request

bluestang2006, 2023-05-28:
This issue is to request required features and extensions that should improve AMDVLK on VKD3D-Proton. Just recently, VKD3D added [VK Profiles](https://github.com/HansKristian-Work/vkd3d-proton/blob/master/VP_D3D12_VKD3D_PROTON_profile.json) for D3D feature and shader model level support.

AFAICT, AMDVLK can support baseline D3D Feature Level (FL) 12.0 and up to Shader Model (SM) 6.0 but doesn't support FL 12.2 and SM 6.6 because of a few missing features/extensions. While some D3D12 games are playable on AMDVLK w/ current drivers (v2023.Q2.2), others are not due to missing features/extensions. 

Lastly, there are a few other features that are needed for optimal performance.

For FL 12.1, AMDVLK will need:
```
VK_EXT_fragment_shader_interlock
    VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT
        fragmentShaderSampleInterlock = true
        fragmentShaderPixelInterlock = true
```
For FL 12.2, AMDVLK will need:
```
VK_EXT_mesh_shader (partially supported)
    VkPhysicalDeviceMeshShaderFeaturesEXT
        taskShader = true

VK_EXT_conservative_rasterization (partially supported)
    VkPhysicalDeviceConservativeRasterizationPropertiesEXT
        fullyCoveredFragmentShaderInputVariable = true
```
For SM 6.6, AMDVLK will need:
```
VK_NV_compute_shader_derivatives
    VkPhysicalDeviceComputeShaderDerivativesFeaturesNV
        computeDerivativeGroupLinear = true
```
For optimal performance w/ FL 12.0, AMDVLK will need:
```
VkPhysicalDeviceLimits
    maxPushConstantsSize = 256
```
For overall optimal performance, AMDVLK will need:
```
VK_EXT_graphics_pipeline_library (ongoing work to support this is in-progress)
VK_EXT_extended_dynamic_state2 (partially supported)
    VkPhysicalDeviceExtendedDynamicState2FeaturesEXT
        extendedDynamicState2PatchControlPoints = true
VK_KHR_present_id
VK_KHR_present_wait
```

oscarbg, 2023-07-01:
nice review!

oscarbg, 2023-10-28:
opened issue asking specifically for taskShader for enable mesh shaders under VKD3D now that games like Alan Wake 2 can make use of it.. (for better perf?)
also you can update the list to include VK_NV_device_generated_commands_compute for Starfield and also support/exposed on RADV under VKD3D..

jinjianrong, 2024-02-26:
@bluestang2006  VK_EXT_graphics_pipeline_library is disabled for dxvk now because of a corruption with VK_EXT_graphics_pipeline_library enabled. This will be fixed in next release.
The VKCTS 1.3.5.2 tests can pass in our testing. What version of vkcts are you using?

bluestang2006, 2024-02-27:
> @bluestang2006 VK_EXT_graphics_pipeline_library is disabled for dxvk now because of a corruption with VK_EXT_graphics_pipeline_library enabled. This will be fixed in next release. The VKCTS 1.3.5.2 tests can pass in our testing. What version of vkcts are you using?

I'm using a much more recent version of VKCTS, off the main branch. I'll wait for the next release when it's enabled for DXVK. Any updates on the Feature Level (FL) requests for VKD3D?

jinjianrong, 2024-03-04:
We reproduced failures in dEQP-VK.pipeline.fast_linked_library.bind_buffers_2.* with new version CTS.  It's pre-existing and has to do with robustness 2 not GPL fast link. We are working on it. 

Regarding the Feature Level (FL) and Shader Model (SM) support for VKD3D: 
•	VK_NV_compute_shader_derivatives
        We implemented the support we care about and the extension is enabled via app detect for steam.
        Discussions will resume to ship the EXT version very soon.  
•	VK_EXT_conservative_rasterization (fullyCoveredFragmentShaderInputVariable)
        We do not plan on supporting this.
•	VK_EXT_fragment_shader_interlock
        We plan to expose it for select apps via app detect.
        We still do not want to enable widely because it's not preferred behavior.
•	VK_EXT_extended_dynamic_state2 (extendedDynamicState2PatchControlPoints)
        We do not plan on supporting this.
•       VkPhysicalDeviceLimits ( maxPushConstantsSize = 256)
        This is a HW limitation for us. We don’t have that many user data registers
•	VK_KHR_present_id
         Currently no plan to implement, but we are checking to put this into future plan
•	VK_KHR_present_wait
         Currently no plan to implement, but we are checking to put this into future plan
•	Task shaders
        We plan to support it in proprietary driver, but no plan to expose it in open source driver yet

Triang3l, 2024-03-04:
> We implemented the support we care about and the extension is enabled via app detect for steam.

> We plan to expose it for select apps via app detect.

Adding workarounds for broken apps that are already in the wild is fine, but when AMDVLK just selectively chooses not to fully support all apps except for some specific ones, that only makes me as an app developer selectively not want to support AMDVLK at all — or support it by faking VkApplicationInfo and possibly detouring other detection mechanisms, which, I think, might actually be a good technical measure to counter this practice. Especially if this filtering is done to show strong sides of the hardware/driver in a few curated examples while sweeping the subpar aspects under the rug, then I don't think it would be respectful to use apps relying on certain features as word-of-mouth marketing demos when in reality you're actively setting up roadblocks for developers who want to use that functionality. This is not the responsibility of a driver, especially in Vulkan which is explicitly designed in a way that lets drivers and apps communicate as much information about each other as possible.

nhaehnle, 2024-03-05:
>  • Task shaders
> We plan to support it in proprietary driver, but no plan to expose it in open source driver yet

Task shader support is open-source. It's part of VK_EXT_mesh_shader, but it's conditional on kernel and firmware support. I've added more info in #341.

jinjianrong, 2024-03-08:
@Triang3l  Thanks for your feedback!
For VK_NV_compute_shader_derivatives, we are selectively enabling it temporarily. The extension is planned for removal when the EXT version of it is available. The EXT spec's release has been delayed because Roadmap releases took priority but should be back on track soon.
 
For VK_EXT_fragment_shader_interlock, we will consider the decision to expose it by default once we have an implementation.

oscarbg, 2024-03-10:
> We reproduced failures in dEQP-VK.pipeline.fast_linked_library.bind_buffers_2.* with new version CTS. It's pre-existing and has to do with robustness 2 not GPL fast link. We are working on it. [...]

@jinjianrong really, many thanks for sharing your plan/roadmap on AMDVLK VKD3D support!! "finally" expectations are clear!

oscarbg, 2024-03-10:
> @Triang3l Thanks for your feedback! For VK_NV_compute_shader_derivatives, we are selectively enabling it temporarily. The extension is planned for removal when the EXT version of it is available. [...]

@jinjianrong @Triang3l a suggestion regarding VK_NV_compute_shader_derivatives and VK_EXT_fragment_shader_interlock exposure.. even if you have concerns enabling unconditionally, excepting selected/curated apps, you have already a mechanism in place via AMDVLK_ENABLE_DEVELOPING_EXT env variable to expose such extensions maybe..
remembered reading https://www.phoronix.com/news/AMDVLK-Developing-Ext-Var 5 years ago..
or maybe this AMDVLK_ENABLE_DEVELOPING_EXT env variable has been removed?

oscarbg, 2024-03-23:
adding latest results of running latest AMDVLK around various VKD3D profiles:

basically primary problems the same but adding new:

> Unsupported extension: VK_EXT_swapchain_maintenance1

Testing profile VP_D3D12_maximum_radv adds:

> Unsupported extension: VK_KHR_shader_maximal_reconvergence

and older ones like:

> Unsupported extension: VK_NV_device_generated_commands
> Unsupported extension: VK_NV_device_generated_commands_compute

```
Testing profile VP_D3D12_FL_11_0_baseline.
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_11_0_baseline, supported: yes
Testing profile VP_D3D12_FL_11_1_baseline.
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_11_1_baseline, supported: yes
Testing profile VP_D3D12_FL_12_0_baseline.
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_12_0_baseline, supported: yes
Testing profile VP_D3D12_FL_12_0_optimal.
Unsupported properties condition: VkPhysicalDeviceProperties2KHR::properties.limits.maxPushConstantsSize >= 256
Unsupported extension: VK_EXT_swapchain_maintenance1
Unsupported extension: VK_KHR_present_id
Unsupported extension: VK_KHR_present_wait
Unsupported feature condition: VkPhysicalDeviceSwapchainMaintenance1FeaturesEXT::swapchainMaintenance1 == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceExtendedDynamicState2FeaturesEXT::extendedDynamicState2PatchControlPoints == VK_TRUE
Unsupported feature condition: VkPhysicalDevicePresentWaitFeaturesKHR::presentWait == VK_TRUE
Unsupported feature condition: VkPhysicalDevicePresentIdFeaturesKHR::presentId == VK_TRUE
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_12_0_optimal, supported: no
Testing profile VP_D3D12_FL_12_1_baseline.
Unsupported extension: VK_EXT_fragment_shader_interlock
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderPixelInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderSampleInterlock == VK_TRUE
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_12_1_baseline, supported: no
Testing profile VP_D3D12_FL_12_2_baseline.
Unsupported extension: VK_EXT_fragment_shader_interlock
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderPixelInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderSampleInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceMeshShaderFeaturesEXT::taskShader == VK_TRUE
Unsupported properties condition: VkPhysicalDeviceConservativeRasterizationPropertiesEXT::fullyCoveredFragmentShaderInputVariable contains VK_TRUE
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_12_2_baseline, supported: no
Testing profile VP_D3D12_FL_12_2_optimal.
Unsupported extension: VK_EXT_fragment_shader_interlock
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderPixelInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderSampleInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceMeshShaderFeaturesEXT::taskShader == VK_TRUE
Unsupported properties condition: VkPhysicalDeviceConservativeRasterizationPropertiesEXT::fullyCoveredFragmentShaderInputVariable contains VK_TRUE
Unsupported extension: VK_EXT_swapchain_maintenance1
Unsupported extension: VK_KHR_present_id
Unsupported extension: VK_KHR_present_wait
Unsupported feature condition: VkPhysicalDeviceSwapchainMaintenance1FeaturesEXT::swapchainMaintenance1 == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceExtendedDynamicState2FeaturesEXT::extendedDynamicState2PatchControlPoints == VK_TRUE
Unsupported feature condition: VkPhysicalDevicePresentWaitFeaturesKHR::presentWait == VK_TRUE
Unsupported feature condition: VkPhysicalDevicePresentIdFeaturesKHR::presentId == VK_TRUE
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_FL_12_2_optimal, supported: no
Testing profile VP_D3D12_maximum_nv.
Unsupported extension: VK_EXT_fragment_shader_interlock
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderPixelInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceFragmentShaderInterlockFeaturesEXT::fragmentShaderSampleInterlock == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceMeshShaderFeaturesEXT::taskShader == VK_TRUE
Unsupported properties condition: VkPhysicalDeviceConservativeRasterizationPropertiesEXT::fullyCoveredFragmentShaderInputVariable contains VK_TRUE
Unsupported extension: VK_KHR_shader_maximal_reconvergence
Unsupported extension: VK_EXT_swapchain_maintenance1
Unsupported extension: VK_KHR_present_id
Unsupported extension: VK_KHR_present_wait
Unsupported feature condition: VkPhysicalDeviceSwapchainMaintenance1FeaturesEXT::swapchainMaintenance1 == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceExtendedDynamicState2FeaturesEXT::extendedDynamicState2PatchControlPoints == VK_TRUE
[... 7 lines cut ...]
Unsupported extension: VK_NVX_image_view_handle
Unsupported extension: VK_NV_device_diagnostic_checkpoints
Unsupported extension: VK_NV_raw_access_chains
Unsupported extension: VK_NV_shader_sm_builtins
Unsupported feature condition: VkPhysicalDeviceRawAccessChainsFeaturesNV::shaderRawAccessChains == VK_TRUE
Unsupported properties condition: VkPhysicalDeviceProperties2KHR::properties.vendorID == 4318
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_maximum_nv, supported: no
[... cut ...]
Unsupported extension: VK_NV_device_generated_commands_compute
Unsupported feature condition: VkPhysicalDeviceDeviceGeneratedCommandsComputeFeaturesNV::deviceGeneratedCompute == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceDeviceGeneratedCommandsFeaturesNV::deviceGeneratedCommands == VK_TRUE
Unsupported feature condition: VkPhysicalDeviceImageCompressionControlFeaturesEXT::imageCompressionControl == VK_TRUE
[RESULT] GPU: AMD Radeon Graphics, Profile: VP_D3D12_maximum_radv, supported: no
```