GPUOpen-Drivers/AMDVLK #325: Visual corruption with water effect in Doom Eternal raytracing

aufkrawall, 2023-06-09:
First open area in first level:

RT on:
[image]

RT off:
[image]

amdvlk 2023.Q2.2-1, RX 6700 XT, 1440p, maximal visuals

bluestang2006, 2023-06-12:
What version of Proton are you using? The only working version for me is Proton 7...
Any version of Proton 8 asserts and refuses the launch the game.

bluestang2006, 2023-09-15:
> First open area in first level:
> 
> RT on: [image]
> 
> RT off: [image]
> 
> amdvlk 2023.Q2.2-1, RX 6700 XT, 1440p, maximal visuals

Looks this has been fixed, I built the dev branch this morning and observed that the water appears correct in the first level.
Keep an eye out for a new release, it should be appearing soon.

closed 2024-05-21 by jinjianrong