GPUOpen-Drivers/AMDVLK #327: Assertion failed: !status && "vkDeferredOperationJoinKHR", file ../src-wine/dlls/winevulkan/loader_thunks.c, line 3128

bluestang2006, 2023-06-17:
System Specs:
CPU - 5900X
GPU - 6800XT w/AMDVLK v2023.Q2.3 driver
OS - Fedora 38 KDE

In Proton 8, some games are hitting the assertion in winevulkan:
`Assertion failed: !status && "vkDeferredOperationJoinKHR", file ../src-wine/dlls/winevulkan/loader_thunks.c, line 3128` which causes the game to abruptly quit.

In my case, Doom Eternal and Portal RTX (TraceRay mode) will hit the assertion consistently. For Doom Eternal, I can fallback to Proton 7 and AMDVLK works fine. For PortalRTX I can only fallback to RayQuery mode for it to work.

I'm not really sure where the regression lies with Proton 8 or AMDVLK. Speaking with Proton devs, they say that earlier versions of Proton were ignoring these fails but with Proton 8 they are no longer ignoring them hence why they are appearing now. They are pointing to AMDVLK as the cause. And FWIW, MESA's RADV driver implements `vkDeferredOperationJoinKHR`  as a nop and it doesn't hit the assert.

*To configure specifc RT modes for PortalRTX the following options in the `rtx.conf` file in the game directory need to be modified:
```
RayQuery[CS]
rtx.renderPassGBufferRaytraceMode = 0
rtx.renderPassIntegrateIndirectRaytraceMode = 0

RayQuery[RGS]
rtx.renderPassGBufferRaytraceMode = 1
rtx.renderPassIntegrateIndirectRaytraceMode = 1

TraceRay[RGS]
rtx.renderPassGBufferRaytraceMode = 2
rtx.renderPassIntegrateIndirectRaytraceMode = 2
```

bluestang2006, 2023-06-28:
Closing this issue, the cause of the issue is Proton 8.

closed 2023-06-28 by bluestang2006

jinjianrong, 2023-07-03:
Refer to https://github.com/ValveSoftware/Proton/issues/3773#issuecomment-1612383500

bluestang2006, 2023-07-04:
Yes, I am aware its a proton issue, that is why I closed the issue 5 days ago.