GPUOpen-Drivers/AMDVLK #328: Assassins Creed Origins White lines

Blisto91, 2023-07-04:
Original dxvk issue reported with the Windows driver: https://github.com/doitsujin/dxvk/issues/3545

When using amdvlk together with dxvk in Assassins Creed Origins in some scenes, like the present day cave scene after prologue, along the edges of objects a white line can appear at certain angles. This does not happen with radv.

<details>
  <summary>Example screenshot</summary>
  
[image]
</details>

Driver version: 2023Q2.3
dxvk version: 2.2 and current master

apitrace with the issue. When i get out of the cave and look inside again the white lines can be seen at the ceiling at certain angles.
https://drive.proton.me/urls/08D98GNA6M#ZZSKXmDXVS1U

CME42, 2023-07-04:
This seems to be related to trunc_coord/truncateCoords, radv disables trunc_coord for textureGather/tg4. I can't repro it anymore with amdvlk when I disable truncateCoords.

There also seems to be a fix for gfx11 with TA_CNTL2.TRUNCATE_COORD_MODE which I can't find in amdvlk: https://gitlab.freedesktop.org/mesa/mesa/-/commit/df6380ddc9c88b37355cb85d2f37697192354aaa

Blisto91, 2023-10-02:
Thanks can confirm it is fixed with 2023.Q3.3

closed 2023-10-02 by Blisto91