GPUOpen-Drivers/AMDVLK #333: Starfield unsupported  graphics card

oldeschool, 2023-09-03:
on Ubuntu 22.04.3 LTS  with an AMD 7900 XT
Starfield fails to start with graphics card does not meet the minimal specification requirements.

runs and plays using RADV -- no start with AMDVLK

oldeschool, 2023-11-21:
problem persists with 2023.Q4.1

JoeLiu-SH, 2023-11-28:
This issue is not caused by Ubuntu or AMD graphics cards. It is because of VKD3D uses the VK_NV_device_ generated_commands function, which is a vender specific extension (not a Khronos standard yet). Khronos has a new version under discussion: https://gitlab.khronos.org/vulkan/vulkan/-/blob/VK_EXT_device_generated_commands/proposals/VK_EXT_device_generated_commands.adoc RADV partially implements VK_NV_device_ generated_commands so it works.

Degerz, 2023-11-29:
@JoeLiu-SH Khronos Group would prefer it's participants to not reveal any information prior to official disclosure ...

JoeLiu-SH, 2023-11-29:
@Degerz Thanks for your reminder, I just updated the comment and removed the link.

Degerz, 2023-11-29:
@JoeLiu-SH Better to delete the comment altogether since it'll show up in edit history ...

JoeLiu-SH, 2023-11-29:
This issue is because of VKD3D uses the VK_NV_device_ generated_commands function, which is vender specific extension (not a Khronos standard yet). RADV partially implements this extension so it works.

jinjianrong, 2024-05-17:
Fixed in 2024.Q2.1

closed 2024-05-17 by jinjianrong