GPUOpen-Drivers/AMDVLK #341: Any reason for not enabling VK_EXT_mesh_shader taskShader feature? (used optionally for Alan Wake2 under VKD3D)

oscarbg, 2023-10-28:
Hi,
thought it was because I'm testing on the Zen4 iGPU (gfx1036 RDNA2).. but seems even the latest big RDNA3 GPUs have it disabled..
RDNA2:
http://vulkan.gpuinfo.org/displayreport.php?id=25275#features_extensions
RDNA3:
http://vulkan.gpuinfo.org/displayreport.php?id=23339#features_extensions

of course RADV exposes it, compare on same RDNA2 hardware:
http://vulkan.gpuinfo.org/compare.php?reports=25275,25276#features_extensions

so question is, are you working/interested on enabling taskShader feature? specially now that work on older GPUs have been removed..

as said it before, in :
https://github.com/GPUOpen-Drivers/AMDVLK/issues/324

without it VKD3D should not expose mesh shaders and without it latest games using mesh shaders like Alan Wake 2 can't make use of it for greater perf..

thanks..

baryluk, 2023-10-29:
Also trying to run amdvlk with Alan Wake 2.  (Mesa works, but there are some minor texture issues when using mesh shaders, and when disabling mesh shaders, it works fine, at 50% fps).

Indeed, taskShader is not advertised:

```
Device Properties and Extensions:
=================================
GPU0:
VkPhysicalDeviceProperties:
---------------------------
        apiVersion        = 1.3.267 (4206859)
        driverVersion     = 2.0.288 (8388896)
        vendorID          = 0x1002
        deviceID          = 0x73bf
        deviceType        = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName        = AMD Radeon RX 6900 XT
        pipelineCacheUUID = 3f93401e-ad4f-5c70-8b23-92bf8ab556b3
...

Device Extensions: count = 159
...
        VK_EXT_mesh_shader                          : extension revision 1
...

VkPhysicalDeviceMeshShaderFeaturesEXT:
--------------------------------------
        taskShader                             = false
        meshShader                             = true
        multiviewMeshShader                    = true
        primitiveFragmentShadingRateMeshShader = true
        meshShaderQueries                      = false

```

Using today's amdvlk `master` repo branch.

baryluk, 2023-10-29:
Also trying to run amdvlk with Alan Wake 2.  (Mesa works, but there are some minor texture issues when using mesh shaders, and when disabling mesh shaders, it works fine, at 50% fps - https://github.com/HansKristian-Work/vkd3d-proton/issues/1754 and https://gitlab.freedesktop.org/mesa/mesa/-/issues/10060  ; at the moment unclear if issue with vkd3d, game, or mesa).

Indeed, taskShader is not advertised:

```
Device Properties and Extensions:
=================================
GPU0:
VkPhysicalDeviceProperties:
---------------------------
        apiVersion        = 1.3.267 (4206859)
        driverVersion     = 2.0.288 (8388896)
        vendorID          = 0x1002
        deviceID          = 0x73bf
        deviceType        = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName        = AMD Radeon RX 6900 XT
        pipelineCacheUUID = 3f93401e-ad4f-5c70-8b23-92bf8ab556b3
...

Device Extensions: count = 159
...
        VK_EXT_mesh_shader                          : extension revision 1
...

VkPhysicalDeviceMeshShaderFeaturesEXT:
--------------------------------------
        taskShader                             = false
        meshShader                             = true
        multiviewMeshShader                    = true
        primitiveFragmentShadingRateMeshShader = true
        meshShaderQueries                      = false

```

Using today's amdvlk `master` repo branch.

baryluk, 2023-10-29:
When starting Alan Wake 2, with this I got a popup with a message:

```
Your GPU does not support all the recommended features. This might lead
to poor performance, visual glitches or crashes. A GPU with the following
additional features is recommended:

 - Mesh Shader
```

The game does start, menu works, and loading works, loading finished, but entering the game, it crashes/exits. I had gdb attached, and unfortunately it is not helpful:

```
(gdb) c
Continuing.

Thread 2 "steam.exe" received signal SIGQUIT, Quit.
[Switching to LWP 2691800]
0x00007fe493f06ee9 in ?? ()
(gdb) bt
#0  0x00007fe493f06ee9 in ?? ()
#1  0x00007fe4931b0db2 in ?? ()
#2  0x0000000000000000 in ?? ()
(gdb) q
A debugging session is active.

	Inferior 1 [process 2691476] will be detached.

Quit anyway? (y or n) y
Detaching from program: /home/user/.config/heroic/tools/proton/Proton-GE-Proton8-21/files/bin/wine64-preloader, process 2691476
[Inferior 1 (process 2691476) detached]
$
```

amdrexu, 2023-11-03:
There is some ongoing work as I know so this feature has not been enabled.

oscarbg, 2023-11-05:
Thanks @amdrexu for info! Also can provide info on if any plans to allow recursive raytracing eventually: https://github.com/GPUOpen-Drivers/AMDVLK/issues/337

amdrexu, 2023-11-05:
> Thanks @amdrexu for info! Also can provide info on if any plans to allow recursive raytracing eventually: #337

Sorry, I don't know the plan. I think there might be some performance concerns.

nhaehnle, 2024-03-05:
Task shaders are supported by AMDVLK, but it is conditional on certain kernel and firmware features. Please see PAL's amdgpuDevice.{h,cpp} for details.

The reason for this is that:

* Task shaders require firmware support, without which they cannot run at all
* Task shaders require gang submit to work reliably in all cases. It may be the case that RADV doesn't have this check and YOLOs it -- in practice it will likely be fine as long as you're not running multiple task shader applications at once. But proper support requires a recent enough kernel driver

oscarbg, 2024-03-10:
@nhaehnle many thanks for clarifying..
based on your comments and seeing the code in amdgpuDevice.cpp (which I reference below), I have two questions:
1) "about getting fwSupportTaskShader = true":
you mention:
> Task shaders require firmware support
and in the code I see, for RDNA2:
> m_chipProperties.pfpUcodeVersion >= 95
and for RDNA3 I see:
> pfpUcodeVersion` >= 1549
I have RDNA2, so how to know if the "pfpUcodeVersion" on my system is good enough i.e >=95 for:
> fwSupportTaskShader = true;
even cloning linux firmware git which has latest amdgpu firmware or even copying from latest AMDGPU Pro driver like
https://www.amd.com/en/support/kb/release-notes/rn-amdgpu-unified-linux-23-40-2-0
are going to be recent enough for task shader support?

2) you mention:
> Task shaders require gang submit to work reliably in all cases. It may be the case that RADV doesn't have this check and YOLOs it -- in practice it will likely be fine as long as you're not running [...]
anyway to forcely enable "supportsGangSubmit" like RADV does.. for people who want to try even if "systems hangs"..
maybe even using/setting some "existing" AMDVLK env variable like AMDVLK_ENABLE_DEVELOPING_EXT=VK_EXT_mesh_shader should disable "gang submit" check and in case firmware recent enough finally enable task shaders.

the code I see:
```
// This code is added here because it is entirely reliant on kernel level support for implicit/explicit
        // gang submit. As a result, this GFXIP-specific logic is being handled in InitQueueInfo.
        const bool supportsGangSubmit = SupportsGangSubmit();

        if (IsGfx103PlusExclusive(m_chipProperties.gfxLevel))
        {
            // Task-shader CTS may fail on Linux upstream stack due to the low FW version.
            bool fwSupportTaskShader = false;

            if (IsGfx103(m_chipProperties.gfxLevel))
            {
                if (m_chipProperties.pfpUcodeVersion >= 95)
                {
                    fwSupportTaskShader = true;
                }
            }
#if PAL_BUILD_GFX11
            else if (IsGfx11(m_chipProperties.gfxLevel))
            {
                if (m_chipProperties.pfpUcodeVersion >= 1549)
                {
                    fwSupportTaskShader = true;
                }
            }
#endif
```

nhaehnle, 2024-03-12:
> I have RDNA2, so how to know if the "pfpUcodeVersion" on my system is good enough i.e >=95 for:

I believe [UMR](https://gitlab.freedesktop.org/tomstdenis/umr/) shows this in GUI mode, or you can build your own driver and look in the debugger? :)

I haven't checked this myself, but the latest linux-firmware really ought to be sufficient.

> anyway there is a way to forcely enable "supportsGangSubmit"?

No. And yeah, it's a fair point that having such a debug/developer override based on an environment variable might be helpful.

jinjianrong, 2024-03-19:
With the basekit from [Radeon™ Software for Linux® 23.40.2 Release Notes | AMD](https://www.amd.com/en/support/kb/release-notes/rn-amdgpu-unified-linux-23-40-2-0) and amdvlk2024.Q1.2, we see the task shaders support is exposed.

oscarbg, 2024-03-20:
> With the basekit from [Radeon™ Software for Linux® 23.40.2 Release Notes | AMD](https://www.amd.com/en/support/kb/release-notes/rn-amdgpu-unified-linux-23-40-2-0) and amdvlk2024.Q1.2, we see the task shaders support is exposed.

thanks for confirmation!!
@jinjianrong can confirm exact kernel version used in testing, and exact GPU? gfx1036 expected to work?

oscarbg, 2024-03-20:
@nhaehnle thanks for info.. still not getting task shaders reported..

> I believe [UMR](https://gitlab.freedesktop.org/tomstdenis/umr/) shows this in GUI mode, or you can build your own driver and look in the debugger? :)
> I haven't checked this myself, but the latest linux-firmware really ought to be sufficient.

 thanks!,  I built latest UMR 1.09 and seeing with Ubuntu noble firmware which currently is up to Februray'24:
[image]
 and with AMD 23.40.2 firmware:
[image]

seeing 
> m_chipProperties.pfpUcodeVersion   = pfpFwVersion;
and from screenshoots seems PFP firmware version is 0xe so well below 95 required..

any hint, help @jinjianrong ..

oscarbg, 2024-03-23:
small update..
overlooked this function where see drm and kernel checking:
```
~/amdvlk/vulkandriver/drivers/pal/src/core/os/amdgpu/amdgpuDevice.h:
bool supportsGangSubmit = ((IsDrmVersionOrGreater(3,49) || IsKernelVersionEqualOrGreater(6,1)) &&
                                    Settings().enableGangSubmit);
```
I see I'm on 6.8 so OK!

anyway I compiled driver by myself and see:
```
AMDLVKDEBUG! supportsGangSubmit 1
AMDLVKDEBUG! gfx103 m_chipProperties.pfpUcodeVersion 14

```
so seems the problem is in firmware as UMRGUI showed:

`m_chipProperties.pfpUcodeVersion 14`

jinjianrong, 2024-03-25:
> exact kernel version used in testing, and exact GPU? gfx1036 expected to work?
OS: ubuntu 22.04.4
Kernel: 6.5.0-25
GPU: gfx1030, navi21 (FW: PFP feature version: 44, firmware version: 0x00000061)
GPU: gfx1100, navi31 (FW: PFP feature version: 29, firmware version: 0x00000660) 
We tried above two GPUs, both can expose task shader support. However, the firmware version check might not be valid for APUs. We need to check it further.

oscarbg, 2024-03-25:
>However, the firmware version check might not be valid for APUs. 
thanks for sharing the info @jinjianrong !
I compiled amdvlk.so skipping the firmware check in the meantime and tested task shaders work ok!, anyway
have a question, can expect the firmware check in APUs propagates also to closed source Vulkan driver in AMDGPU PRO right? and also even to Windows Vulkan driver?
I say because both Windows and Linux "AMDGPU Pro" AMDVLK latest drivers still also list task shaders unsupported on gfx1036..
and assuming Windows driver integrates latest firmware..
thanks..

jinjianrong, 2024-05-16:
The ucode version check for task shader enablement is revised in 2024.Q2.1

closed 2024-05-16 by jinjianrong