GPUOpen-Drivers/AMDVLK #342: Failed to compile gpurt pipelines: Common.hlsl:553:18: error: firstbithigh is not yet implemented for 64-bit width components when targetting SPIR-V

baryluk, 2023-10-29:
gcc version 13.2.0 (Debian 13.2.0-6) 

```
user@debian:~/amdvlk$ dxc --version
libdxcompiler.so: 1.7(dev;4190-64030a4e)
```

DirectXShaderCompiler from Today's main branch at 64030a4e01e27c608b6c5f1ab5add43575e086f9

I know, README.md says to use tagged earlier release, but it has issues with gcc 13.2.

```
user@debian:~$ ./amdvlk-build.sh 
+ cd /home/user
+ mkdir -p /home/user/amdvlk
+ cd /home/user/amdvlk
+ repo init --partial-clone --depth=1 -u https://github.com/GPUOpen-Drivers/AMDVLK.git -b master
/home/user/amdvlk/.repo/repo/main.py:569: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn("\n... A new version of repo (%s) is available.", exp_str)

... A new version of repo (2.37) is available.
/home/user/amdvlk/.repo/repo/main.py:580: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn(
... New version is available at: /home/user/amdvlk/.repo/repo/repo
... The launcher is run from: /usr/bin/repo
!!! The launcher is not writable.  Please talk to your sysadmin or distro
!!! to get an update installed.

repo: reusing existing repo client checkout in /home/user/amdvlk

repo has been initialized in /home/user/amdvlk
++ nproc
+ repo sync --no-clone-bundle --no-tags -j32
/home/user/amdvlk/.repo/repo/main.py:569: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn("\n... A new version of repo (%s) is available.", exp_str)

... A new version of repo (2.37) is available.
/home/user/amdvlk/.repo/repo/main.py:580: DeprecationWarning: The 'warn' method is deprecated, use 'warning' instead
  logger.warn(
... New version is available at: /home/user/amdvlk/.repo/repo/repo
... The launcher is run from: /usr/bin/repo
!!! The launcher is not writable.  Please talk to your sysadmin or distro
!!! to get an update installed.

remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0        
Fetching: 100% (8/8), done in 0.099s
repo sync has finished successfully.
+ cmake -GNinja -Sdrivers/xgl/ -Bdrivers/xgl/builds/Release64
CMake Warning at cmake/XglPackaging.cmake:32 (message):
  PACKAGE_NAME is not specified, default is amdvlk
Call Stack (most recent call first):
  CMakeLists.txt:56 (include)

CMake Warning at cmake/XglPackaging.cmake:37 (message):
  PACKAGE_VERSION is not specified, default is 1.0
Call Stack (most recent call first):
  CMakeLists.txt:56 (include)

CMake Warning (dev) at CMakeLists.txt:220 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /home/user/amdvlk/drivers/third_party/metrohash/CMakeLists.txt:8 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
[... 400 lines cut ...]
                     ^
In file included from /home/user/amdvlk/drivers/gpurt/src/shaders/BuildBVH.hlsl:45:
In file included from /home/user/amdvlk/drivers/gpurt/src/shaders/Common.hlsl:47:
In file included from /home/user/amdvlk/drivers/gpurt/src/shaders/Debug.hlsl:30:
/home/user/amdvlk/drivers/gpurt/src/shaders/Extensions.hlsl:202:89: warning: parameter 'timerLo' is uninitialized when used here [-Wparameter-usage]
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                                                        ^~~~~~~~~~~~~~~
/home/user/amdvlk/drivers/gpurt/src/shaders/../shared/rayTracingDefs.h:62:29: note: expanded from macro 'DUMMY_VOID_FUNC'
#define DUMMY_VOID_FUNC   { }
                            ^
/home/user/amdvlk/drivers/gpurt/src/shaders/Extensions.hlsl:202:64: note: variable 'timerLo' is declared here
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                               ^
/home/user/amdvlk/drivers/gpurt/src/shaders/Common.hlsl:41:22: note: expanded from macro 'out_param'
#define out_param(x) out x
                     ^
/home/user/amdvlk/drivers/gpurt/src/shaders/Common.hlsl:553:18: error: firstbithigh is not yet implemented for 64-bit width components when targetting SPIR-V
    return (63 - firstbithigh(value));
                 ^
Failed to compile Vulkan shader config ShaderConfig< Path: BuildBVH.hlsl, EntryPoint: BuildBVH, OutputName: None, BaseLogicalId: None, RootSignaturePath: None, Defines: GPURT_BUILD_RTIP2=1,GPURT_DEVELOPER=1,GPURT_CLIENT_API_VULKAN=1,GPURT_CLIENT_INTERFACE_MAJOR_VERSION=39,USE_BUILD_LBVH=1,GPURT_BVH_BUILD_SHADER=1, GroupTag: BVH >
Launching threads for raytracing shader compilation...
Raytracing shader compilation completed with no errors.
Launching threads for raytracing shader compilation...
ninja: build stopped: cannot make progress due to previous errors.
```

baryluk, 2023-10-29:
Using a workaround for gcc 13.2, for that old tagged release which I described in the dxc issue https://github.com/microsoft/DirectXShaderCompiler/issues/5949

patching one file for missing include (`#include <cstdint>` in `DirectXShaderCompiler/tools/clang/include/clang/Basic/Version.h`, and adding extra cmake flags (`-DCMAKE_CXX_FLAGS="-Wno-error=stringop-overflow -Wno-changes-meaning"`) for the compiler to ignore some warnings / errors, I compiled this older dxc.

Then amdvlk does build:

```
[1/148] Generating pipelines/g_internal_shaders.h, pipelines/g_GpuRtLibrary.h
Launching threads for raytracing shader compilation...
Raytracing shader compilation completed with no errors.
Launching threads for raytracing shader compilation...
Raytracing shader compilation completed with no errors.
...
```

So, this looks like an issue of amdvlk xgl with newer dxc.

mareksapota, 2024-04-06:
I've run into a similar issue trying to compile the latest AMDVLK release.  Code compiles with an older version of DirectXShaderCompiler but the latest version fails with `error: firstbithigh is not yet implemented for 64-bit width components when targetting SPIR-V`.

Versions used:
amdvlk v-2024.Q1.3
glslang 14.1.0
xgl b7100c18eb2d134098738f97d7cc3a26e485862f
pal 42e29f4d3d528c6a59c2db77fb6d70127fb9ffab
llpc 14535c028dc70a093d95a95efedb749812c1e77d
llvm-project da55e9e39bc7868016b7b265aa535dff84b7d591
MetroHash 18893fb28601bb9af1154cd1a671a121fff6d8d3
CWPack 4f8cf0584442a91d829d269158567d7ed926f026
gpurt 2e6d528610f45a99723d6e4f15f24da12fd6a1e5

Working DirectXShaderCompiler version:
DirectXShaderCompiler v1.7.2308

Not working DirectXShaderCompiler version:
DirectXShaderCompiler v1.8.2403.2

qiaojbao, 2024-04-07:
Also ran into this build issue with VulkanSDK1.3.275+'s DXC tools.
We are working on it.

qiaojbao, 2024-04-08:
could you help to try PR-8 please?
https://github.com/GPUOpen-Drivers/gpurt/pull/8

mareksapota, 2024-04-09:
I added your patch to my RPM build and was able to get a successful build with DirectXShaderCompiler v1.8.2403.2.

https://build.opensuse.org/package/show/home:mareksapota:branches:X11:XOrg/amdvlk

qiaojbao, 2024-04-10:
Fix merged.

closed 2024-04-10 by qiaojbao