GPUOpen-Drivers/AMDVLK #344: What is the purpose of this driver?

xanscale, 2023-11-27:
What is the purpose of this driver? Why don't you just join efforts with the Mesa team? why would anyone use this driver instead of the Mesa one that comes pre-installed on all distros?

If there's a reason to keep things separate, I'd just like to know.

Thank you

Flakebi, 2023-11-27:
Thanks for your interest in our driver.
The amdvlk code base exists separately from mesa because the code is shared between Linux and Windows.
A lot of the code is actually also shared with other (non-Vulkan) drivers.

Mesa also shares its code between drivers and vendors but it is not used for Windows drivers.
&nbsp;

If amdvlk or radv is better usually depends on the concrete use case/game. For some games radv performs better, for other games amdvlk performs better.
Vulkan makes it easy to install both and choose between them by changing an environment variable.

I hope I was able to answer your questions.

xanscale, 2023-11-27:
hi, thanks for your answer

I still think this is not the best way to spend your software budget on linux drivers, considering that most linux users will use radv

Degerz, 2023-11-29:
I believe that this driver is also being used by an AMD customer on Android as well so it isn't just about sharing code between Linux and Windows anymore ?

jinjianrong, 2023-11-29:
AMDVLK is mainly to provide reference implementation to our customers, as well as RADV (such as new ASIC support).

closed 2023-11-29 by jinjianrong