GPUOpen-Drivers/AMDVLK #348: AMDVLK fails to initialize when an unsupported/legacy GPU is present

farnoy, 2024-01-10:
Updating from 2023.Q3 to Q4, I'm seeing the following issue:

```
$ VK_ICD_FILENAMES=/nix/store/h72hnivwggayn0825wxmdy1i29i732ha-amdvlk-2023.Q3.3/share/vulkan/icd.d vulkaninfo --summary
'DISPLAY' environment variable not set... skipping surface info
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.268

Instance Extensions: count = 19
-------------------------------
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_portability_enumeration         : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6
VK_LUNARG_direct_driver_loading        : extension revision 1

Instance Layers: count = 2
--------------------------
VK_LAYER_KHRONOS_validation Khronos Validation Layer     1.3.268  version 1
VK_LAYER_MESA_device_select Linux device selection layer 1.3.211  version 1

Devices:
========
GPU0:
        apiVersion         = 1.3.264
        driverVersion      = 2.0.285
        vendorID           = 0x1002
        deviceID           = 0x66af
        deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName         = AMD Radeon VII
        driverID           = DRIVER_ID_AMD_OPEN_SOURCE
        driverName         = AMD open-source driver
        driverInfo         = 2023.Q3.3 (LLPC)
        conformanceVersion = 1.3.3.1
        deviceUUID         = 00000000-4300-0000-0000-000000000000
        driverUUID         = 414d442d-4c49-4e55-582d-445256000000
GPU1:
        apiVersion         = 1.3.264
        driverVersion      = 2.0.285
        vendorID           = 0x1002
        deviceID           = 0x73bf
        deviceType         = PHYSICAL_DEVICE_TYPE_DISCRETE_GPU
        deviceName         = AMD Radeon RX 6900 XT
        driverID           = DRIVER_ID_AMD_OPEN_SOURCE
        driverName         = AMD open-source driver
        driverInfo         = 2023.Q3.3 (LLPC)
        conformanceVersion = 1.3.3.1
        deviceUUID         = 00000000-4600-0000-0000-000000000000
        driverUUID         = 414d442d-4c49-4e55-582d-445256000000
```

```
$ VK_ICD_FILENAMES=/nix/store/ayib2fdv2jvqjv48s4yzx2aq8c4lnyzk-amdvlk-2023.Q4.2/share/vulkan/icd.d/amd_icd64.json vulkaninfo --summary
[... 475 lines cut ...]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/ga152zm3lr7y43987pgafaw2mmklc8lp-libXau-1.0.11/lib/libXau.so.6 [0]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/szrdvlc8nhdyv2nfz59fp29wld7jnvlk-libXdmcp-1.1.4/lib/libXdmcp.so.6 [0]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/libpthread.so.0 [0]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/55rql61j6rppix4vhck8fxcwxsrp6spn-libffi-3.4.4/lib/libffi.so.8 [0]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/libc.so.6 [0]
   3165108:
   3165108:
   3165108:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/ld-linux-x86-64.so.2 [0]
   3165108:
```
</details>

My guess would be that it's related to the Radeon VII that lost support with the Q4 release

cc @Flakebi (sorry about the ping if it's not relevant to you)

farnoy, 2024-01-10:
Couldn't post the same output for Q3 as the comment was too long:

<details>
<summary>here it is</summary>

```
$ VK_ICD_FILENAMES=/nix/store/h72hnivwggayn0825wxmdy1i29i732ha-amdvlk-2023.Q3.3/share/vulkan/icd.d LD_DEBUG=libs vulkaninfo --summary
   3165526:     find library=libdl.so.2 [0]; searching
   3165526:      search path=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/glibc-hwcaps/x86-64-v3:/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/glibc-hwcaps/x86-64-v2:/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib               (RUNPATH from file vulkaninfo)
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/libdl.so.2
   3165526:      search path=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib:/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib/glibc-hwcaps/x86-64-v3:/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib/glibc-hwcaps/x86-64-v2:/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib:/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/glibc-hwcaps/x86-64-v3:/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/glibc-hwcaps/x86-64-v2:/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib:/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib/glibc-hwcaps/x86-64-v3:/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib/glibc-hwcaps/x86-64-v2:/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib:/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib/glibc-hwcaps/x86-64-v3:/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib/glibc-hwcaps/x86-64-v2:/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib            (RUNPATH from file vulkaninfo)
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/libdl.so.2
   3165526:       trying file=/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/wsq5wqcwy92ailryz8vx4y26fjx7dgp0-vulkan-tools-1.3.268.0/lib/libdl.so.2
   3165526:       trying file=/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/libdl.so.2
   3165526:       trying file=/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib/libdl.so.2
   3165526:       trying file=/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib/libdl.so.2
   3165526:      search path=/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/glibc-hwcaps/x86-64-v3:/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/glibc-hwcaps/x86-64-v2:/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib             (system search path)
   3165526:       trying file=/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/glibc-hwcaps/x86-64-v3/libdl.so.2
   3165526:       trying file=/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/glibc-hwcaps/x86-64-v2/libdl.so.2
   3165526:       trying file=/nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/libdl.so.2
   3165526:
   3165526:     find library=libxcb.so.1 [0]; searching
   3165526:      search path=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib            (RUNPATH from file vulkaninfo)
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/libxcb.so.1
   3165526:      search path=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib:/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib:/nix/store/b6lkm9rqb8s4254q1fnrvk80qdkxasym-libX11-1.8.7/lib:/nix/store/6aqnbq957vs2i8q0cwg5an0pqgwldrfz-wayland-1.22.0/lib             (RUNPATH from file vulkaninfo)
   3165526:       trying file=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib/libxcb.so.1
   3165526:       trying file=/nix/store/jglzsibhzn9is8mm63z47m2dg1dm7kq3-libxcb-1.16/lib/libxcb.so.1
   3165526:
   3165526:     find library=libX11.so.6 [0]; searching
   3165526:      search path=/nix/store/mnw61wpgp088pfdri08giwc72agbnrjx-vulkan-loader-1.3.268.0/lib            (RUNPATH from file vulkaninfo)
[... cut ...]
   3165526:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/libpthread.so.0 [0]
   3165526:
   3165526:
   3165526:     calling fini: /nix/store/55rql61j6rppix4vhck8fxcwxsrp6spn-libffi-3.4.4/lib/libffi.so.8 [0]
   3165526:
   3165526:
   3165526:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/libc.so.6 [0]
   3165526:
   3165526:
   3165526:     calling fini: /nix/store/jm7y1aw31kfll35p3hr8cin2wrn13sx0-glibc-2.38-27/lib/ld-linux-x86-64.so.2 [0]
   3165526:
```
</details>

JakeMerdichAMD, 2024-01-11:
Hi @farnoy. This has been fixed internally but hasn't made it to a release yet. If you want a quick local fix, change [this line](https://github.com/GPUOpen-Drivers/pal/blob/6353b182d28fe85f28b40d1809291b515fcddd3e/src/core/os/amdgpu/amdgpuDevice.cpp#L213) to `Result::Unsupported`.

jinjianrong, 2024-01-26:
Please check the fix with 2024.Q1.1

closed 2024-01-26 by jinjianrong