GPUOpen-Drivers/AMDVLK #349: Low performance and graphical issues with AMDVLK in Detroit Become Human-compared to RADV

Leopard1907, 2024-01-12:
Tested drivers:

- Mesa-git as of today
- AMDVLK 2023.Q4.3 (LLPC)

Hardware:

- 7900 XTX gpu
- 7800X3D cpu

There are often graphical issues with AMDVLK when playing this game on both Wine and Proton Experimental.

Captured some rdc captures which exposes the issue. Renderdoc 1.30 was used.

Such as:

[image]
[image]
[image]

Link to those rdc captures ( It would be nice to notify me after you download them so i can delete as 7.7 gb is a rather hefty space being wasted ) :

https://drive.google.com/file/d/1h_QhU1pwKDidqxngAQiXfXyaEaNTIaV-/view?usp=sharing

Also here is the perf comparison between AMDVLK and RADV in the chapter called "A New Home":

[image]
[image]

Approx 70 FPS diff.

Settings used were verything set to ultra, 1440p resolution.

Game only has 30 and 60 fps options by default but that limit can be changed via this.

https://www.pcgamingwiki.com/wiki/Detroit:_Become_Human#High_frame_rate

If my save files etc are needed ( EGS version ) you can use this:

[INTER_1.tar.gz](https://github.com/GPUOpen-Drivers/AMDVLK/files/13924153/INTER_1.tar.gz)

Untar it into `drive_c/users/beast/Saved Games/Quantic Dream/Detroit Become Human` inside the prefix. That naming might be INTER_3 for Steam copies.

perlfu, 2024-01-15:
@Leopard1907 I have downloaded your RenderDoc captures, so you can clear them from your Google Drive if you need to free space.

I am able to reproduce the visual corruption using the RenderDoc captures; however, I do not see the same corruption in the game.  It's possible that this is a driver/game interaction issue which may already have been resolved.  I will attempt to reproduce with the exact driver you used.

I can reproduce the AMDVLK frame rate for the scene shown, but I have yet to get the game to work with the Mesa version specified.  It seems to stall during the "Shader Compilation Process".

Leopard1907, 2024-01-15:
@perlfu Weird, just compiled mesa-git freshly againts this https://gitlab.freedesktop.org/mesa/mesa/-/commit/480ec5b1d480c62bc9984a44bbb6453bca1970be and it still goes through shader compile screen.

[image]

Leopard1907, 2024-01-15:
@perlfu Did you compile RADV as debug build, i compiled as release.

ishitatsuyuki, 2024-01-15:
As leopard suggested, there seems to be an issue with debug builds of Mesa; I don't know exactly why, but release or debugoptimized builds should let the game go through shader compilation without problems.

If that still doesn't work, please try compiling this branch: https://gitlab.freedesktop.org/ishitatsuyuki/mesa/-/tree/more-dbh-hack. The branch contains a targeted workaround for the fact that the game refuses to consider any pipeline cache below 128KB to be "finished".

perlfu, 2024-01-16:
Thanks for the pointers, I had completely forgotten default mesa build is debug.
I can get the game running with a release build.

One important reproduction detail is that the game resets graphics settings when switching ICD.  So game has to be launched once to build pipelines, before setting resolution and editing .ini file to unlock frame rate.

For LLPC I got ~95 fps, with 85% GPU, 35% CPU.
For RADV I got ~125fps, with 70% GPU, 50% CPU.

My test CPU is older and slower than your test system, which I think explains the RADV difference between our results.
I think RADV is CPU bound in this test.  I never saw RADV use more than 70% GPU.
This seems to be a strong reproduction of a performance gap from my perspective.

Memory usage is definitely higher in LLPC, particularly in the pipeline compilation phase of game launch there seems to be some possible leakage.

Leopard1907, 2024-01-16:
Thanks for the test. Windows side seems to be also suffering from perf issues but with potentially much less cpu usage. But ofc i don't have Windows to test.

https://youtu.be/5IBmkPMQMUA?si=rjC-tDQoOTIORm1x

There is just an insane difference here.

Leopard1907, 2024-01-16:
> Thanks for the test. Windows side seems to be also suffering from perf issues but with potentially much less cpu usage. But ofc i don't have Windows to test.
> 
> https://youtu.be/5IBmkPMQMUA?si=rjC-tDQoOTIORm1x
> 
> There is just an insane difference here.

Btw my screenshots/testing also shows much higher cpu usage on Radv side. 75-80 percent on all cores.

jinjianrong, 2024-05-21:
The quality issues are already fixed in new release

Leopard1907, 2024-11-14:
@jinjianrong Is there any update about performance?

jinjianrong, 2024-11-19:
We have identified the pipelines that cause the low performance, working on optimization in llvm.

Leopard1907, 2025-09-10:
@jinjianrong @perlfu Any news about this? Since more than a year passed, i wonder what happened.

Leopard1907, 2025-09-15:
Closing per https://github.com/GPUOpen-Drivers/AMDVLK/discussions/416.

Mission completed, RADV prevailed.

closed 2025-09-15 by Leopard1907