GPUOpen-Drivers/AMDVLK #351: wayland: use linux-dmabuf instead of wl_drm

emersion, 2024-01-20:
amdvlk relies on the `wl_drm` Wayland protocol to find the DRM device used by the compositor. However, `wl_drm` is a Mesa-private, legacy protocol, and compositors are [starting to drop support](https://github.com/swaywm/sway/issues/7897). amdvlk should use the standard linux-dmabuf protocol from wayland-protocols instead.

Flakebi, 2024-01-25:
Thanks for the heads-up. This is tracked in https://github.com/GPUOpen-Drivers/pal/issues/96.

CNR0706, 2024-01-26:
Would implementing linux-dmabuf also allow Tearing on Wayland? Or is this unrelated?

Not super experienced with this. Sorry if that's a dumb question.

jinjianrong, 2024-05-17:
Done in 2024.Q4.1 release

mikezackles, 2024-06-16:
In case anyone else comes across this and does a double-take, it's in the [2024.Q2.1 release notes](https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2024.Q2.1), so I'm guessing Q4 was a typo. Anyway, thanks!

jinjianrong, 2024-06-17:
Sorry, this is a typo. The work is done in 2024.Q2.1 release

HurricanePootis, 2024-06-30:
I am running AMDVLK 2024.Q2.3 and I am still getting the following error whenever I try to run a wayland application using AMDVLK:

```
wl_drm#44: error 0: wl_drm.create_prime_buffer is not implemented
```

jinjianrong, 2024-07-01:
Thanks for reporting the issue. We will check it further.

lukelmy, 2024-07-02:
Hi @HurricanePootis, thanks for your report.
Could you please provide your wayland application or where I can download it?

HurricanePootis, 2024-07-03:
@lukelmy 

I am running `vkcube-wayland`, which is apart of the [Vulkan-Tools](https://github.com/KhronosGroup/Vulkan-Tools) repository. 

Also, for more information, I am running the `amdvlk` driver with the following build process:
```
  cmake -H. -Bbuilds/Release64 \
    -DCMAKE_BUILD_TYPE=Release \
    -DBUILD_WAYLAND_SUPPORT=On \
    -G Ninja
```
And, according to the `default.xml` file in the tar soure code, it is built with PAL commit https://github.com/GPUOpen-Drivers/pal/commit/695337aa8131913d883ff15787cd75f6fcb11229.

HurricanePootis, 2024-07-28:
Any update on this? Kind of inconvenient for amdvlk not to work with modern wayland apps.

jinjianrong, 2024-08-06:
@HurricanePootis we can't reproduce the issue locally. But we do notice an issue that may be related with yours. A fix is under implementation

HurricanePootis, 2024-08-07:
@jinjianrong Good to know. If you are curious, I installed amdvlk through archlinux and [here](https://gitlab.archlinux.org/archlinux/packaging/packages/amdvlk/-/blob/main/PKGBUILD?ref_type=heads).

HurricanePootis, 2024-08-07:
I am looking at the files pulled by the archlinux `amdvlk` package during its build process of 2024.Q2.3, and at `pal/src/core/os/amdgpu/wayland/waylandWindowSystem.h`, `zwp_linux_dmbauf` does show up and `pal/src/core/os/wayland/mesa/` does exist.

So then, I don't know why even though these files are present at build time, the ensuing amdvlk driver is trying to use wl_drm.

olm52, 2024-09-12:
Same problem here with uo.to.date Archlinux using

`ENABLE_HDR_WSI=1 mpv --vo=gpu-next --target-colorspace-hint --gpu-api=vulkan --gpu-context=waylandvk '/home/audiolinux/Downloads/Temp/Life Untouched 4K Demo.mp4' `

`[audiolinux@archlinux ~]$ ENABLE_HDR_WSI=1 mpv --vo=gpu-next --target-colorspace-hint --gpu-api=vulkan --gpu-context=waylandvk '/home/audiolinux/Downloads/Temp/Life Untouched 4K Demo.mp4' 
 (+) Video --vid=1 (*) (hevc 3840x2160 59.940fps)
 (+) Audio --aid=1 (*) (aac 2ch 48000Hz)
[HDR Layer] Created HDR surface
AO: [pipewire] 48000Hz stereo 2ch floatp
VO: [gpu-next] 3840x2160 yuv420p10
[HDR Layer] Creating swapchain for id: 5 - format: VK_FORMAT_B8G8R8A8_UNORM - colorspace: VK_COLOR_SPACE_SRGB_NONLINEAR_KHR
wl_drm#56: error 0: wl_drm.create_prime_buffer is not implemented
[vo/gpu-next/wayland] Error occurred on the display fd
Exiting... (Quit)
`

HurricanePootis, 2024-09-12:
Sounds like an Arch Issue? I will make a gitlab issue on their [tracker](https://gitlab.archlinux.org/archlinux/packaging/packages/amdvlk).

lf-, 2024-10-04:
https://gitlab.archlinux.org/archlinux/packaging/packages/amdvlk/-/issues/1 arch bug filed

HurricanePootis, 2024-10-06:
awesome

jinjianrong, 2024-11-21:
@HurricanePootis we had a fix in pal dev branch: 
https://github.com/GPUOpen-Drivers/pal/commit/758683bc4c64d56287a1beb8384d19d94251200f
https://github.com/GPUOpen-Drivers/pal/commit/7b182c9310b3ed8eb98f2239c36a008ffb3eea3c

Could you help check if this fixes the issue?

HurricanePootis, 2024-11-22:
@jinjianrong I just tested out `vkcube-wayland` and an SDL2 game, and was able to run them with Wayland with AMDVLK 2024.Q4.1 with these two patches applied, thank you so much!

HurricanePootis, 2024-12-05:
I compiled 2024.Q4.2, and got vkcube_wayland working

closed 2024-12-06 by jinjianrong