GPUOpen-Drivers/AMDVLK #354: Corruption with Nanite foliage and shadows in The Talos Principle 2

aufkrawall, 2024-02-23:
There is flickering corruption going on with foliage and shadows in The Talos Principle 2. Should be instantly noticeable after loading the first level and walking around a bit:

[image]

This is probably related to Nanite (which can't be toggled in that game).

amdvlk [v-2024.Q1.2](https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2024.Q1.2)
Radeon RX 7800 XT
Proton Experimental

jinjianrong, 2024-05-17:
Not reproduced with 2024.Q2.1 release

aufkrawall, 2024-05-17:
It is even more broken than before, in a very obvious manner:
[image]

jinjianrong, 2024-06-05:
The new issue is a regression. This is fixed 2024.Q2.2 release

closed 2024-06-05 by jinjianrong

aufkrawall, 2024-06-05:
Well, yes, indeed the foliage looks fixed now. But big level geometry parts are still broken with 2024.Q2.2 on the 7800 XT:
[image]