GPUOpen-Drivers/AMDVLK #355: Nvidia Remix issues (Windows)

KingVulpes, 2024-03-08:
Tested on Windows 11 with a RX 7600 at 1080 with TAA-U performance 
1. On 7000 series, using volumetric lighting cause visual glitches
2. When using newer versions of Nvidia Remix Runtime, it'll show a failed to create 3d3 error which wasn't an issue before on 23.12.1, this prevents games like Portal Prelude RTX from launching
3. In Portal RTX, after 23.12.1, when playing through test chambers 18 and 19, frame rate will drop from 40fps to below 20fps and sometimes is just unplayable 
4. In Portal RTX on all driver versions the game will crash on the end credits
5. The Remix toolkit refuses to load any Remix runtime captures

6000 series issue
As we all know, Remix runs horrible on RDNA2, but this seems to only be a Windows issue as Linux is completely playable and bug free, don't know if there's anything that can be done about this

jinjianrong, 2024-08-21:
The issue is no more reproducible with latest driver

jinjianrong, 2024-08-21:
If you still run into issue with Windows driver, please report issue on https://github.com/GPUOpen-Drivers/AMD-Gfx-Drivers/issues