GPUOpen-Drivers/AMDVLK #357: Compiling fails on Musl-libC - error with NULL type;

SpidFightFR, 2024-04-03:
Hello, while compiling amdvlk from source for the Void-Linux distribution, the maintainers (thank you Johnnynator on the Void repo for your precious help !) noticed a small issue while compiling with musl-libc.

In the file `pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp b/pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp`:

at lines `10155` and `10167`, the usage of `NULL,` makes an error as it's interpreted as `std::nullptr_t` instead of a `long unsigned integer` as usually expected.

The GlibC compiler seems kinda tolerant to this, as the compiling process goes through without issue.
however Musl stops right through.

According to Johnny's help and my tests, replacing `NULL,` by `0UL,` in both places, seems to fix the issue.

Just in case, we both also made a git patch for our Voidlinux compiler, it will be linked to this issue, for convenience.

<details><summary>amdvlk-fix-musl-build.patch</summary>

```patch
diff --git a/pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp b/pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp
index 2d62ec3ef..e14ccbb9c 100644
--- a/pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp
+++ b/pal/src/core/hw/gfxip/gfx9/gfx9UniversalCmdBuffer.cpp
@@ -10152,7 +10152,7 @@ uint32 UniversalCmdBuffer::BuildExecuteIndirectIb2Packets(
                                                                                      drawIndexReg,
                                                                                      1,
                                                                                      pParamData[cmdIndex].argBufSize,
-                                                                                     NULL,
+                                                                                     0UL,
                                                                                      PacketPredicate(),
                                                                                      usesLegacyMsFastLaunch,
                                                                                      pDeCmdIb2Space);
@@ -10164,7 +10164,7 @@ uint32 UniversalCmdBuffer::BuildExecuteIndirectIb2Packets(
                                                                                       drawIndexReg,
                                                                                       1,
                                                                                       pParamData[cmdIndex].argBufSize,
-                                                                                      NULL,
+                                                                                      0UL,
                                                                                       PacketPredicate(),
                                                                                       usesLegacyMsFastLaunch,
                                                                                       pDeCmdIb2Space);

```
</details>

Best regards - SpidFightFR

qiaojbao, 2024-04-10:
Thanks, I will make a PR for this.

qiaojbao, 2024-04-10:
https://github.com/GPUOpen-Drivers/pal/pull/98

closed 2024-04-10 by SpidFightFR