GPUOpen-Drivers/AMDVLK #361: [feature request] Implement VK_EXT_depth_bias_control

Blisto91, 2024-04-25:
VK_EXT_depth_bias_control is, when avaliable, used by dxvk to solve some differences in how depth bias is handled in Vulkan versus Direct3D which can otherwise lead to graphical issues in some games. It would be nice if amdvlk could implement this extension for more complete dxvk support.

For further reference see also dxvk implementation PR https://github.com/doitsujin/dxvk/pull/3501 and depth bias mega-issue https://github.com/doitsujin/dxvk/pull/3501

jinjianrong, 2024-05-21:
The extension is being implemented

Blisto91, 2025-01-14:
Friendly bump. Is this still being worked on?

jinjianrong, 2025-01-15:
We finally decided to drop this extension.

closed 2025-01-15 by jinjianrong

Triang3l, 2025-01-15:
> We finally decided to drop this extension.

Is that due to the `lowZplanePolyOffsetBits` part that introduces scaling of the bias by 4 for D24_UNORM and by 2 for D16_UNORM?

https://github.com/GPUOpen-Drivers/pal/blob/c7fc61289a6eae6c87458fe362f88a44c0c53900/src/core/hw/gfxip/gfx9/gfx9DepthStencilView.cpp#L271

`NEG_NUM_DB_BITS = -22` used with `lowZplanePolyOffsetBits` already appears to be not conformant with the specification of the depth bias for `D24_UNORM`, which says:

"For fixed-point depth attachment representations […] value is implementation-dependent but must be at most `r = 2 × 2^-n` where n is the number of bits used for the depth aspect when using a fixed-point attachment, or the number of mantissa bits plus one when using a floating-point attachment."

The hardware (at least according to what I remember from my tests on the ancient GFX4, but I wouldn't expect changes in this area) appears to be calculating `r` as `2^NEG_NUM_DB_BITS` for `DB_IS_FLOAT_FMT = 0`. So, in order to stay within the `r = 2 × 2^-n` limit for D24_FLOAT, for which `n` is 24, `NEG_NUM_DB_BITS` can't be larger than -23.

Overall, this extension seems to be trivial to implement as:

```
if (representation == FLOAT) {
  // Like in Direct3D 9, already exposed as `absoluteDepthBias` in PAL.
  DB_IS_FLOAT_FMT = 0;
  NEG_NUM_DB_BITS = 0;
} else {
  if (format == D32_FLOAT) {
    if (representation == LEAST_REPRESENTABLE_VALUE_FORCE_UNORM) {
      DB_IS_FLOAT_FMT = 0;
      // n = number of explicit mantissa bits plus one.
      NEG_NUM_DB_BITS = -24;
    } else {
      DB_IS_FLOAT_FMT = 1;
      NEG_NUM_DB_BITS = -23;
    }
  } else if (format == D24_UNORM) {
    DB_IS_FLOAT_FMT = 0;
    NEG_NUM_DB_BITS = -24;
  } else if (format == D16_UNORM) {
    DB_IS_FLOAT_FMT = 0;
    NEG_NUM_DB_BITS = -16;
  }

  if (DB_IS_FLOAT_FMT == 0) {
    if (depthBiasExact) {
      // The Vulkan specification requires r = 2^-n.
      // The hardware does r = 2^NEG_NUM_DB_BITS, exactly what's needed by the specification.
    } else {
      // For unorm formats, r = 2^-n is not enough to ensure separation between two same primitives,
      // because the smallest representable difference is 1/(2^n - 1) rather than 1/(2^n).
      // Make the bias r = 2 × 2^-n, which is within the limit allowed by the specification for `depthBiasExact` disabled.
      ++NEG_NUM_DB_BITS;
    }
  }
}
```

misyltoad, 2025-01-15:
> We finally decided to drop this extension.

Disappointing, this extension is incredibly trivial to implement and basically required for DXVK D3D9 to function correctly in some titles.

jinjianrong, 2025-01-22:
The effort is actually bigger that we imaged. we had initial implementation for the extension, but found VK_EXT_dynamic_rendering  did not working with depth bias control. it requires some restructuring in Pal that also impacts other clients