GPUOpen-Drivers/AMDVLK #363: RenderDoc Capture with Mesh shader payload causes GPU resets and system freezes

Firestar99, 2024-05-16:
Opening a RenderDoc Capture that has task and mesh shaders which utilize a payload to send data between them causes GPU resets and system freezes. I have found two very different ways of triggering it, both of them somehow related to amdvlk. These repo instructions assume a clean Ubuntu 24.04 system to start, so only RADV and no AMDVLK installed.

## Standard AMDVLK Capture
1. Install [AMDVLK](https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2024.Q2.1) deb package on your system
2. Open the AMDVLK [capture](https://github.com/GPUOpen-Drivers/AMDVLK/files/15339600/new_amdvlk_freeze.zip)
3. Rarely the RenderDoc will freeze here already
4. Select the single `vkCmdDrawMeshTasksEXT` call
5. Observe the RenderDoc window freezing, and a 5/5 system freeze up to ~1min later

## Opening a RADV Capture while AMDVLK is *just* present but unused
1. Open the **RADV** [capture](https://github.com/GPUOpen-Drivers/AMDVLK/files/15339586/new_radv_with_amdvlk_installed_gpu_reset.zip) and observe Renderdoc working as expected
2. Install [AMDVLK](https://github.com/GPUOpen-Drivers/AMDVLK/releases/tag/v-2024.Q2.1) deb package on your system
3. Delete `/etc/vulkan/implicit_layer.d/amd_icd64.json` to remove the `VK_LAYER_AMD_switchable_graphics_64` implicit layer, which forces you to always use the amdvlk driver
4. verify that `vulkanCapsViewer` can see both drivers, RADV with `AMD Radeon Graphics (RADV REMBRANDT)` and amdvlk with `AMD Radeon Graphics` (I wish amdvlk had a more identifiable name)
5. Open the same **RADV** [capture](https://github.com/GPUOpen-Drivers/AMDVLK/files/15339586/new_radv_with_amdvlk_installed_gpu_reset.zip) again, but this time observe Renderdoc freezing, likely followed by a 5/5 GPU reset or rarely 1/5 system freeze
6. With RADV you don't even need to select the draw itself, loading the capture is almost always enough.

=> My current conclusion is that an amdvlk device being available, even though it is unused, is enough to cause the Renderdoc to freeze and a gpu reset to follow. I verified this by simply removing the icd json.

## Related issues
https://github.com/baldurk/renderdoc/issues/3309
https://gitlab.freedesktop.org/mesa/mesa/-/issues/11156

WenqingLiAMD, 2024-07-09:
Hello @Firestar99 ,
Thanks for the bug report.

I tried to reproduce the issue with `new_amdvlk_freeze.rdc` and `new_radv_with_amdvlk_installed_gpu_reset.rdc` on NAVI21, but neither GPU resets nor system freezes are observerd.

> => My current conclusion is that opening a RADV capture and an amdvlk device being available, even though it is unused, is enough to cause the Renderdoc to freeze and a gpu reset to follow.

Can you please confirm above conclusion by export VK_DRIVER_FILES to force the loader to use radv _or_ amdvlk? For example, if the env is not set, we can see both radv and amdvlk devices:

```
INFO | DRIVER:    linux_read_sorted_physical_devices:
INFO | DRIVER:         Original order:
INFO | DRIVER:               [0] AMD Radeon RX 6800 (RADV NAVI21)
INFO | DRIVER:               [1] llvmpipe (LLVM 15.0.7, 256 bits)
INFO | DRIVER:               [2] AMD Radeon RX 6800
INFO | DRIVER:         Sorted order:
INFO | DRIVER:               [0] AMD Radeon RX 6800 (RADV NAVI21)
INFO | DRIVER:               [1] AMD Radeon RX 6800
INFO | DRIVER:               [2] llvmpipe (LLVM 15.0.7, 256 bits)
```

once you specify the radv json: export VK_DRIVER_FILES=/usr/share/vulkan/icd.d/radeon_icd.x86_64.json, you can only see radv device
```
INFO | DRIVER:    linux_read_sorted_physical_devices:
INFO | DRIVER:         Original order:
INFO | DRIVER:               [0] AMD Radeon RX 6800 (RADV NAVI21)
INFO | DRIVER:         Sorted order:
INFO | DRIVER:               [0] AMD Radeon RX 6800 (RADV NAVI21)
```

jinjianrong, 2024-07-25:
Please re-open if the issue is still reproducible

closed 2024-07-25 by jinjianrong

Firestar99, 2024-09-02:
Finally got some time to do additional testing. As it's been a while, I decided to grab the newest release `v-2024.Q3.2` of 4 days ago, and am still able to reproduce everything as described with ease. 

(more testing to follow here...)

Based on current results, I would assume this could be something hardware specific, as I'm testing on a 680M iGPU of my Ryzen 9 6900HS. vulkaninfo: [RADV](http://vulkan.gpuinfo.org/displayreport.php?id=32241) [AMDVLK](http://vulkan.gpuinfo.org/displayreport.php?id=32982)

jinjianrong, 2024-09-06:
We  still can't reproduce the issue on REMBRANDT.
For “Opening a RADV Capture while AMDVLK is just present but unused” case, it is known that Vulkan loader doesn't work correctly when both amdvlk and RADV are presenting. This is the reason we introduced the  AMD_switchable_graphics layer.

Firestar99, 2024-09-09:
I'm honestly not sure how to further proceed, with how simple it is for me to replicate it, but you failing to to do so. Back when I first reported it, I also tried doing a clean install of Kubuntu 24.04 (Ubuntu with KDE Plasma Desktop), and only when I installed amdvlk and removed your layer, it started crashing when I ran / debugged my demo.

Maybe there's some difference in the versions used for testing, which would help explain the repo difficulties?
Ubuntu 24.04
KDE Plasma Wayland
kernel 6.8.0-41-generic
mesa / radv 24.0.9-0ubuntu0.1
amdvlk 2024.Q3.2

Wrt. removing your layer: The initial reason I installed amdvlk was wanting to play around with RGP on my demo. Personally, I wanted to keep RADV  as my main driver and when profiling or testing, explicitly select the amdvlk driver. So your layer was more of a hindrance on that workflow, which is why I removed it.

jinjianrong, 2024-09-10:
" only when I installed amdvlk and removed your layer, it started crashing when I ran / debugged my demo."
This is expected. When both amdvlk and RADV are installed, the AMD_switchable_graphics layer is needed to switch the driver between AMDVLK and RADV by environment variable AMD_VULKAN_ICD = AMDVLK or RADV.
If you don't want to use the AMD_switchable_graphics layer, you need to disable one of the driver JSON files, either amdvlk or radv.

felixaszx, 2025-07-30:
The same issue is also observed on Windows AMDVLK driver on RX9060xt