GPUOpen-Drivers/AMDVLK #367: Compiling fails on v-2024-Q2-3

SpidFightFR, 2024-06-26:
Here are my error logs:

<details><summary>Logs</summary>

```
[2542/2843] Building interface/lgc/LgcDialect.h.inc...
FAILED: compiler/llpc/llvm/tools/lgc/interface/lgc/LgcDialect.h.inc /builddir/amdvlk-2024.Q2.3/xgl/build/compiler/llpc/llvm/tools/lgc/interface/lgc/LgcDialect.h.inc 
cd /builddir/amdvlk-2024.Q2.3/xgl/build && /builddir/amdvlk-2024.Q2.3/xgl/build/compiler/llpc/llvm/bin/llvm-dialects-tblgen -gen-dialect-decls --dialect lgc -I /builddir/amdvlk-2024.Q2.3/llpc/lgc -I/builddir/amdvlk-2024.Q2.3/xgl/build/compiler/llpc/llvm/include -I/builddir/amdvlk-2024.Q2.3/llvm-project/llvm/include -I/builddir/amdvlk-2024.Q2.3/llpc/lgc/../imported/llvm-dialects/include /builddir/amdvlk-2024.Q2.3/llpc/lgc/interface/lgc/LgcDialect.td --write-if-changed -o compiler/llpc/llvm/tools/lgc/interface/lgc/LgcDialect.h.inc -d compiler/llpc/llvm/tools/lgc/interface/lgc/LgcDialect.h.inc.d
/builddir/amdvlk-2024.Q2.3/llpc/lgc/interface/lgc/LgcDialect.td:166:24: error: Variable not defined: 'ImmutableStringAttr'
  let arguments = (ins ImmutableStringAttr:$format, varargs:$args);
                       ^
[2543/2843] Building LgcRtqDialect.cpp.inc...
[2544/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/csv_reporter.cc.o
[2545/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/perf_counters.cc.o
[2546/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/reporter.cc.o
[2547/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/json_reporter.cc.o
[2548/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/benchmark.cc.o
[2549/2843] Building CXX object compiler/llpc/llvm/tools/CompilerUtils/CMakeFiles/LLVMCompilerUtils.dir/lib/CompilerUtils.cpp.o
/builddir/amdvlk-2024.Q2.3/llpc/compilerutils/lib/CompilerUtils.cpp: In member function 'llvm::iterator_range<llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::BasicBlock, false, false, void, false>, false, false> > CompilerUtils::CrossModuleInliner::inlineCall(llvm::CallBase&)':
/builddir/amdvlk-2024.Q2.3/llpc/compilerutils/lib/CompilerUtils.cpp:255:16: warning: unused variable 'bbCount' [-Wunused-variable]
  255 |   const size_t bbCount = targetFunc->size();
      |                ^~~~~~~
[2550/2843] Building CXX object compiler/llpc/llvm/third-party/benchmark/src/CMakeFiles/benchmark.dir/statistics.cc.o
[2551/2843] Building CXX object compiler/llpc/llvm/tools/CompilerUtils/CMakeFiles/LLVMCompilerUtils.dir/lib/TypeLowering.cpp.o
[2552/2843] Building CXX object compiler/llpc/llvm/lib/Target/AMDGPU/MCTargetDesc/CMakeFiles/LLVMAMDGPUDesc.dir/AMDGPUMCCodeEmitter.cpp.o
[2553/2843] Building CXX object compiler/llpc/llvm/lib/Target/AMDGPU/Disassembler/CMakeFiles/LLVMAMDGPUDisassembler.dir/AMDGPUDisassembler.cpp.o
ninja: build stopped: subcommand failed.

```
</details>

any idea what's causing this ?

jinjianrong, 2024-06-27:
Please check if the imported/llvm-dialects submodule is update correctly.

SpidFightFR, 2024-06-27:
Hey, thanks for your answer, indeed i was using an out of date version of llvm-dialect ! i now updated to the latest master commit (`ed4b46e8425066a96a5e79afc29bce3d82eecf71`) and it now seems to work.

Thanks again !
(also it would be extra cool to have the used commit in `default.xml` or something.)

closed 2024-06-27 by SpidFightFR

jinjianrong, 2024-06-28:
llvm-dialect is a submodule of LLPC. The commit used is specified in imported/llvm-dialects. I guess you didn't recursively sync the submodule when you update llpc.