GPUOpen-Drivers/AMDVLK #369: Support for VK_KHR_display_swapchain

theoparis, 2024-07-13:
See https://registry.khronos.org/vulkan/specs/1.3-extensions/man/html/VK_KHR_display_swapchain.html

Unfortunately it seems like mesa does not implement this in radeonsi either 😕 
This is a blocker for my 3d wayland compositor I have been trying to implement...

KenSZ-AMD, 2024-07-31:
Hi, may I know what is your use case exactly (i.e., tools, games, etc.)? Are you working on Virtual Reality sort of things?

jinjianrong, 2024-07-31:
@theoparis  could you use VK_EXT_physical_device_drm for your purpose, just like Sway/wlroots does (
[Missing VK_EXT_physical_device_drm · Issue #306 · GPUOpen-Drivers/AMDVLK · GitHub](https://github.com/GPUOpen-Drivers/AMDVLK/issues/306)) ?