GPUOpen-Drivers/AMDVLK #375: AMDVLK disables DCC based on feedback loop image usage instead of based on the image layout

DadSchoorse, 2024-09-03:
Relevant XGL code: https://github.com/GPUOpen-Drivers/xgl/blob/de3316323e16b2b71a738efc7312f254069f28a6/icd/api/vk_image.cpp#L539-L544

DXVK creates all d3d9 images that can be used as an attachment with that usage. Then, when a feedback loop is detected, it changes the image layout to `VK_IMAGE_LAYOUT_ATTACHMENT_FEEDBACK_LOOP_OPTIMAL_EXT`. 

Disabling delta color compression based on the image usage means that DCC will always be disabled, even for games that don't use feedback loops. This will result in unnecessary and potentially large performance decrease. Instead, AMDVLK should disable dcc when the application transitions to the feedback loop image layout.

kd-11, 2024-09-04:
RPCS3 does the same thing. If feedback loop layout is supported the usage flag is set for all render targets and the layout is switched on-demand.

jinjianrong, 2024-12-09:
Fixed in 2024.Q4.2

closed 2024-12-09 by jinjianrong