GPUOpen-Drivers/AMDVLK #377: Silent Hill 2 Remake and Alan Wake 2 issues with AMDVLK and PRO drivers

serhii-nakon, 2024-10-19:
It hard to describe everything here so I will put here link to issue on VKD3D-Proton where I created ticket for this problem - https://github.com/HansKristian-Work/vkd3d-proton/issues/2166

In general when use RADV games works even with RT (but slowly).
When I tried switch to PRO - Silent Hill 2 Remake disabled RT and no ability to enable it due no switcher for it at all. Alan Wake 2 crash kernel driver and freeze game.
When I tried switch to AMDVLK - Silent Hill 2 Remake able to enable RT but it crash the game. Alan Wake 2 crash kernel driver and freeze game.

Game crash
[image]

Kernel log
```
[ 5657.482864] amdgpu 0000:03:00.0: amdgpu: ring gfx_0.0.0 timeout, signaled seq=301213, emitted seq=301215
[ 5657.484082] amdgpu 0000:03:00.0: amdgpu: Process information: process AlanWake2.exe pid 14644 thread vkd3d_queue pid 14711
[ 5657.484084] amdgpu 0000:03:00.0: amdgpu: GPU reset begin!
[ 5659.542085] amdgpu 0000:03:00.0: amdgpu: MES failed to respond to msg=REMOVE_QUEUE
[ 5659.542089] [drm:amdgpu_mes_unmap_legacy_queue [amdgpu]] *ERROR* failed to unmap legacy queue
[ 5659.729856] [drm:gfx_v11_0_hw_fini [amdgpu]] *ERROR* failed to halt cp gfx
[ 5659.777094] amdgpu 0000:03:00.0: amdgpu: Dumping IP State
[ 5659.779170] amdgpu 0000:03:00.0: amdgpu: Dumping IP State Completed
[ 5659.779172] amdgpu 0000:03:00.0: amdgpu: MODE1 reset
[ 5659.779174] amdgpu 0000:03:00.0: amdgpu: GPU mode1 reset
[ 5659.779222] amdgpu 0000:03:00.0: amdgpu: GPU smu mode1 reset
[ 5660.284301] amdgpu 0000:03:00.0: amdgpu: GPU reset succeeded, trying to resume
[ 5660.284491] [drm] PCIE GART of 512M enabled (table at 0x00000085FEB00000).
[ 5660.284557] [drm] VRAM is lost due to GPU reset!
[ 5660.284558] amdgpu 0000:03:00.0: amdgpu: PSP is resuming...
[ 5660.354062] amdgpu 0000:03:00.0: amdgpu: reserve 0x1300000 from 0x85fc000000 for PSP TMR
[ 5660.498576] amdgpu 0000:03:00.0: amdgpu: RAP: optional rap ta ucode is not available
[ 5660.498578] amdgpu 0000:03:00.0: amdgpu: SECUREDISPLAY: securedisplay ta ucode is not available
[ 5660.498580] amdgpu 0000:03:00.0: amdgpu: SMU is resuming...
[ 5660.498582] amdgpu 0000:03:00.0: amdgpu: smu driver if version = 0x0000003d, smu fw if version = 0x00000040, smu fw program = 0, smu fw version = 0x004e7e00 (78.126.0)
[ 5660.498585] amdgpu 0000:03:00.0: amdgpu: SMU driver if version not matched
[ 5660.654563] amdgpu 0000:03:00.0: amdgpu: SMU is resumed successfully!
[ 5660.656587] [drm] DMUB hardware initialized: version=0x07002A00
[ 5660.677288] amdgpu 0000:03:00.0: amdgpu: ring gfx_0.0.0 uses VM inv eng 0 on hub 0
[ 5660.677290] amdgpu 0000:03:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 1 on hub 0
[ 5660.677292] amdgpu 0000:03:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 4 on hub 0
[ 5660.677293] amdgpu 0000:03:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 6 on hub 0
[ 5660.677294] amdgpu 0000:03:00.0: amdgpu: ring comp_1.3.0 uses VM inv eng 7 on hub 0
[ 5660.677295] amdgpu 0000:03:00.0: amdgpu: ring comp_1.0.1 uses VM inv eng 8 on hub 0
[ 5660.677295] amdgpu 0000:03:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 9 on hub 0
[ 5660.677296] amdgpu 0000:03:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 10 on hub 0
[ 5660.677297] amdgpu 0000:03:00.0: amdgpu: ring comp_1.3.1 uses VM inv eng 11 on hub 0
[ 5660.677298] amdgpu 0000:03:00.0: amdgpu: ring sdma0 uses VM inv eng 12 on hub 0
[ 5660.677299] amdgpu 0000:03:00.0: amdgpu: ring sdma1 uses VM inv eng 13 on hub 0
[ 5660.677300] amdgpu 0000:03:00.0: amdgpu: ring vcn_unified_0 uses VM inv eng 0 on hub 8
[ 5660.677301] amdgpu 0000:03:00.0: amdgpu: ring vcn_unified_1 uses VM inv eng 1 on hub 8
[ 5660.677302] amdgpu 0000:03:00.0: amdgpu: ring jpeg_dec uses VM inv eng 4 on hub 8
[ 5660.677303] amdgpu 0000:03:00.0: amdgpu: ring mes_kiq_3.1.0 uses VM inv eng 14 on hub 0
[ 5660.690404] amdgpu 0000:03:00.0: amdgpu: recover vram bo from shadow start
[ 5660.702633] amdgpu 0000:03:00.0: amdgpu: recover vram bo from shadow done
[ 5660.702654] amdgpu 0000:03:00.0: amdgpu: GPU reset(2) succeeded!
[ 5660.703743] [drm:amdgpu_cs_ioctl [amdgpu]] *ERROR* Failed to initialize parser -125!
```

I noticed that in VKD3d logs enabled features for:
PRO version it show only about DXR;
AMDVLK it show about DXR, DXR 1.1;
RADV it show about DXR, DXR 1.1 and Ultimate 

Please looks all logs in related ticket and it more detailed.

serhii-nakon, 2024-11-02:
@jinjianrong Hello, I noticed that new release fixed AW2 and it allow to run SH2 Remake (But with artifacts and some freezes and crashes)

jinjianrong, 2024-11-03:
Thanks for the update. we are working on the SH2 issue

jinjianrong, 2025-02-07:
Silent Hill 2 issue is fixed in 2025.Q1.1 release

closed 2025-02-07 by jinjianrong

serhii-nakon, 2025-03-06:
@jinjianrong Hello, I just checked with SH2 Remake and got this - no matter with rt or without the same error when trying to start game

[image]