GPUOpen-Drivers/AMDVLK #379: [vkd3d-proton][feature request] Support VK_EXT_device_generated_commands

Blisto91, 2024-10-29:
Now that `VK_EXT_device_generated_commands` have been released as a proper cross vendor extension, instead of the NV only version, I hope you will consider implementing it for usage with vkd3d-proton.

Best regards

simondeschenes, 2025-02-24:
We are looking at using this extension too. We'd like the flexibility of switching shaders in execution sets and switching pipelines between execution sets.

jsnjkl, 2025-05-16:
Dgc support requires FW update in additional to AMDVLK code itself. Would you share more background of requesting Dgc over AMDVLK? If there's certain busniess value and strategy reason, it may expedite the process.

simondeschenes, 2025-06-11:
I am not sure I understand the need for a FW update. Isn't AMDVLK using the same `amdgpu` firmware linux is using?
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/amdgpu

I also think it is possible for you to take inspiration from the Mesa implementation which is licensed under MIT.
https://gitlab.freedesktop.org/mesa/mesa/-/blob/main/src/amd/vulkan/radv_dgc.c

Of course, it is not going to be pluggable as-is, but I am not sure you want to reinvent the wheel either.