GPUOpen-Drivers/AMDVLK #385: ERROR:             loader_scanned_icd_add: Attempt to retrieve either 'vkGetInstanceProcAddr' or 'vk_icdGetInstanceProcAddr' from ICD /usr/lib/x86_64-linux-gnu/amdvlk64.so failed

82157402, 2024-12-10:
OS :Ubunt 22.04
[image]

Build CMD
1)cmake -G Ninja -S xgl -B builds/Release64 -DCMAKE_C_FLAGS="-fno-lto" -DCMAKE_CXX_FLAGS="-fno-lto"
2)cmake --build builds/Release64

[image]

"vk_icdGetInstanceProcAddrSG" is not setted to the export function.
Why ?

82157402, 2024-12-10:
[vulkaninfo_result_enable_VK_LOADER_DEBUG.txt](https://github.com/user-attachments/files/18074688/vulkaninfo_result_enable_VK_LOADER_DEBUG.txt)

[image]

qiaojbao, 2024-12-10:
could you try  -DVKI_ENABLE_LTO=OFF cmake option?

82157402, 2024-12-10:
cmake -G Ninja -S xgl -B builds/Debug64  -DCMAKE_BUILD_TYPE=Debug -DLLVM_PARALLEL_LINK_JOBS=2  -DVKI_ENABLE_LTO=OFF
I try Debug Version , it is running.

closed 2024-12-10 by 82157402