GPUOpen-Drivers/AMDVLK #389: Euro Truck Simulator 2 ( Linux build, not Windows/Proton ) has visual problems on AMDVLK while RADV renders correctly

Leopard1907, 2024-12-22:
**System details:**

- 7900 XTX
- AMDVLK 2024.Q4.2
- Mesa 24.3.2 and 24.3.2, both are fine

**Issue:**

Euro Truck Simulator 2 has an experimental Vulkan renderer on both Windows and Linux, which can be enabled with game arg on Steam game launch options:

`-rdevice vk`

Renderdoc captures will be provided below.

**Screenshots ( RADV looks fine):** 

- RADV:

[image]

- AMDVLK:

[image]
[image]

**Renderdoc captures:**

https://drive.google.com/file/d/1c2W2KndEKJtPfTa54YEW1H4s4uTXi0E1/view?usp=sharing

https://drive.google.com/file/d/14F9Unr2byFOwDa5Y_81ht5yt9Ju0adSB/view?usp=sharing

Reger95, 2024-12-26:
Hi @Leopard1907,
How did you capture this frame using renderdoc? I also want to capture some frame by renderdoc, but failed.
I've already registered renderdoc on my machine, and when I typed `export ENABLE_VULKAN_RENDERDOC_CAPTURE=1` in shell and execute "vkcube", I can success capture a frame.
Here's what I've done in steam for this game:
1. Click Euro Track Simulator2 in my steam library.
2. Open the setting for this game.
3. Type `-rdevice vk export ENABLE_VULKAN_RENDERDOC_CAPTURE=1` in launch options.
4. Finally click PLAY and I can't launch the game.

Leopard1907, 2024-12-26:
In Steam launch options of the game:

```ENABLE_VULKAN_RENDERDOC_CAPTURE=1 %command% -rdevice vk```

Game arg should be the last thing one passes.