GPUOpen-Drivers/AMDVLK #393: Corrupted Vulkan windows and Gnome File Manager with 2024.Q4.2+

kmshanah, 2025-01-01:
When updating from amdvlk 2024.Q4.1 to amdvlk 2024.Q4.2 (same issue persists with 2024.Q4.3), I started having problems with corrupted gnome file manager window and also simple vulkan applications (for example, vkgears is busted).  The rest of the desktop and my usual applications (terminal, firefox) work fine.

[image]

On a related note, Gnome File manager still doesn't look 100% right with 2024.Q4.1, but it is usable:

[image]

System is running Arch Linux:
linux 6.12.7.arch1-1
linux-firmware 20241210.b00a7f7e-1
mesa 1:24.3.2-1

The system is a Asus G15 AMD Advantage laptop (G513QY) which has both integrated (Ryzen 9 5900HX, Vega 8 GPU) and discrete (RX 6800M) GPUs.  The applications all appear to be using the discrete GPU (e.g. according to `vkgears -info`).

HurricanePootis, 2025-01-23:
This is a known issue upstream with GTK I believe. I think you can force GTK apps to use an alternative renderer besides Vulkan in the mean time.

kmshanah, 2025-01-26:
I've tracked this down to the PAL setting `UseZwpDmaBufProtocol`.
The default was changed from false in 2024.Q4.1 to true in 2024.Q4.2.
When I set this to false, the corruption shown in the first screenshot above does not occur.

daftaupe, 2025-02-02:
> I've tracked this down to the PAL setting `UseZwpDmaBufProtocol`. The default was changed from false in 2024.Q4.1 to true in 2024.Q4.2. When I set this to false via the settings file, the corruption shown in the first screenshot above does not occur with the newer versions.

I can confirm that using 2024.Q4.3, and putting the followingent content into `/etc/amd/amdVulkanSettings.cfg`, I get almost rid of the window corruptions.
```
UseZwpDmaBufProtocol,0
```
Especially the black border as seen in the second capture can be mitigated by expanding the window to fill the screen.

jinjianrong, 2025-02-07:
The black border issue is fixed with 2025.Q1.1 release, please check it.

kmshanah, 2025-02-07:
2025.Q1.1 fixes the black border issue, but seems to be missing the m_presentOnSameGpu fix so the windows are corrupted for me.

jinjianrong, 2025-02-08:
The m_presentOnSameGpu fix caused a performance regression so is not included in this release. it will be in next release

qiaojbao, 2025-02-08:
The latest fix has been ported into dev branch.
https://github.com/GPUOpen-Drivers/pal/commit/32f57f372172874365f6564c92fae65ef91eb29d
@kmshanah , could you help to verify it, please?

kmshanah, 2025-02-08:
> The latest fix has been ported into dev branch.
> [GPUOpen-Drivers/pal@32f57f3](https://github.com/GPUOpen-Drivers/pal/commit/32f57f372172874365f6564c92fae65ef91eb29d)
> [@kmshanah](https://github.com/kmshanah) , could you help to verify it, please?

Yes, can confirm this works for me 👍

closed 2025-03-26 by jinjianrong