GPUOpen-Drivers/AMDVLK #397: AMDVLK 2025.Q1.1 Fails to Compile on Arch Linux

HurricanePootis, 2025-02-18:
# System Information
OS: Arch Linux
glibc version: 2.41+r6+gcf88351b685d-1
gcc version: 14.2.1+r753+g1cd744a6828f
directx-shader-compiler version: 1.8.2407-1

# Error
Whenever compiling amdvlk 2025.Q1.1 in a clean chroot, I get the following errors:
```
FAILED: gpurt/src/pipelines/g_internal_shaders.h gpurt/src/pipelines/g_GpuRtLibrary.h /build/amdvlk/src/xgl/builds/Release64/gpurt/src/pipelines/g_internal_shaders.h /build/amdvlk/src/xgl/builds/Release64/gpurt/src/pipelines/g_GpuRtLibrary.h 
cd /build/amdvlk/src/xgl/builds/Release64/gpurt/src && /usr/bin/python3.13 /build/amdvlk/src/gpurt/tools/CompileRTShaders.py --vulkan --spirv --outputDir /build/amdvlk/src/xgl/builds/Release64/gpurt/src/pipelines --defines "GPURT_DEVELOPER=1;GPURT_CLIENT_API_VULKAN=1;GPURT_CLIENT_INTERFACE_MAJOR_VERSION=41" --includePaths "" --whiteListPath /build/amdvlk/src/gpurt/tools/strip_whitelist.txt /build/amdvlk/src/gpurt/src/shaders/ --strict
Compilation failed for shader EncodeTriangleNodes
Output log:
Compiling EncodeNodes.hlsl:EncodeTriangleNodes -> EncodeTriangleNodes_spv.h...
Subprocess invocation failed. Error code 5. Additional details follow:
Environment: {'SHELL': '/bin/bash', 'DEBUG_FFLAGS': ' -ffile-prefix-map=/build/amdvlk/src=/usr/src/debug/amdvlk', 'BUILDTOOL': 'devtools', 'SUDO_GID': '0', 'PYTHONHASHSEED': '0', 'TEXTDOMAINDIR': '/usr/share/locale', 'SUDO_COMMAND': '/bin/bash -c bash -c cd\\ /startdir;\\ makepkg\\ "$@" -bash --syncdeps --noconfirm --log --holdver --skipinteg --install', 'SUDO_USER': 'root', 'FFLAGS': '  -ffile-prefix-map=/build/amdvlk/src=/usr/src/debug/amdvlk', 'PWD': '/build/amdvlk/src/xgl/builds/Release64/gpurt/src', 'SOURCE_DATE_EPOCH': '1739842254', 'LOGNAME': 'builduser', 'CXXFLAGS': '-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/build/amdvlk/src=/usr/src/debug/amdvlk', 'SUDO_HOME': '/root', 'DEBUG_RUSTFLAGS': '-C debuginfo=2 --remap-path-prefix=/build/amdvlk/src=/usr/src/debug/amdvlk', 'COMMAND_MODE': 'legacy', 'LDFLAGS': '-Wl,-O1 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now          -Wl,-z,pack-relative-relocs', 'HOME': '/build', 'LANG': 'C.UTF-8', 'RUSTFLAGS': '-Cforce-frame-pointers=yes -C debuginfo=2 --remap-path-prefix=/build/amdvlk/src=/usr/src/debug/amdvlk', 'FCFLAGS': '  -ffile-prefix-map=/build/amdvlk/src=/usr/src/debug/amdvlk', 'MAKEFLAGS': '-j16', 'TERM': 'xterm-256color', 'USER': 'builduser', 'SHLVL': '1', 'DEBUGINFOD_URLS': 'https://debuginfod.archlinux.org ', 'CHOST': 'x86_64-pc-linux-gnu', 'PATH': '/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl', 'CFLAGS': '-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -g -ffile-prefix-map=/build/amdvlk/src=/usr/src/debug/amdvlk', 'SUDO_UID': '0', 'MAIL': '/var/mail/builduser', 'BUILDTOOLVER': '1:1.3.1-1-any', 'OLDPWD': '/build/amdvlk/src/xgl/builds/Release64', 'TEXTDOMAIN': 'pacman-scripts', '_': '/usr/bin/python3.13', 'LD_LIBRARY_PATH': ''}
Command arguments: ['dxc', '-I', '', '-fspv-target-env=vulkan1.1', '-spirv', '-E', 'EncodeTriangleNodes', '-DAMD_VULKAN', '-DAMD_VULKAN_DXC', '-DAMD_VULKAN_SPV', '-DGPURT_DEVELOPER=1', '-DGPURT_CLIENT_API_VULKAN=1', '-DGPURT_CLIENT_INTERFACE_MAJOR_VERSION=41', '-DGPURT_BVH_BUILD_SHADER=1', '-fvk-use-scalar-layout', '-Od', '-Vd', '-T', 'cs_6_2', '-Wno-ignored-attributes', '-Wno-parentheses-equality', '-Fo', 'EncodeTriangleNodes.spv', '/build/amdvlk/src/gpurt/src/shaders/EncodeNodes.hlsl']
Expected empty output: False
Working directory: /build/amdvlk/src/xgl/builds/Release64/gpurt/src/pipelines/spv/EncodeTriangleNodes
Stdout:

Stderr:
In file included from /build/amdvlk/src/gpurt/src/shaders/EncodeNodes.hlsl:55:
In file included from /build/amdvlk/src/gpurt/src/shaders/Common.hlsl:50:
In file included from /build/amdvlk/src/gpurt/src/shaders/Debug.hlsl:30:
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:160:3: warning: parameter 'status' is uninitialized when used here [-Wparameter-usage]
) DUMMY_VOID_FUNC
  ^~~~~~~~~~~~~~~
/build/amdvlk/src/gpurt/src/shaders/../shared/rayTracingDefs.h:65:29: note: expanded from macro 'DUMMY_VOID_FUNC'
#define DUMMY_VOID_FUNC   { }
                            ^
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:159:5: note: variable 'status' is declared here
    out_param(uint)    status       ///< Hit status
    ^
/build/amdvlk/src/gpurt/src/shaders/Common.hlsl:44:22: note: expanded from macro 'out_param'
#define out_param(x) out x
                     ^
In file included from /build/amdvlk/src/gpurt/src/shaders/EncodeNodes.hlsl:55:
In file included from /build/amdvlk/src/gpurt/src/shaders/Common.hlsl:50:
In file included from /build/amdvlk/src/gpurt/src/shaders/Debug.hlsl:30:
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:89: warning: parameter 'timerHi' is uninitialized when used here [-Wparameter-usage]
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                                                        ^~~~~~~~~~~~~~~
/build/amdvlk/src/gpurt/src/shaders/../shared/rayTracingDefs.h:65:29: note: expanded from macro 'DUMMY_VOID_FUNC'
#define DUMMY_VOID_FUNC   { }
                            ^
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:39: note: variable 'timerHi' is declared here
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                      ^
/build/amdvlk/src/gpurt/src/shaders/Common.hlsl:44:22: note: expanded from macro 'out_param'
#define out_param(x) out x
                     ^
In file included from /build/amdvlk/src/gpurt/src/shaders/EncodeNodes.hlsl:55:
In file included from /build/amdvlk/src/gpurt/src/shaders/Common.hlsl:50:
In file included from /build/amdvlk/src/gpurt/src/shaders/Debug.hlsl:30:
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:89: warning: parameter 'timerLo' is uninitialized when used here [-Wparameter-usage]
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                                                        ^~~~~~~~~~~~~~~
/build/amdvlk/src/gpurt/src/shaders/../shared/rayTracingDefs.h:65:29: note: expanded from macro 'DUMMY_VOID_FUNC'
#define DUMMY_VOID_FUNC   { }
                            ^
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:64: note: variable 'timerLo' is declared here
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                               ^
/build/amdvlk/src/gpurt/src/shaders/Common.hlsl:44:22: note: expanded from macro 'out_param'
#define out_param(x) out x
                     ^
In file included from /build/amdvlk/src/gpurt/src/shaders/EncodeNodes.hlsl:59:
In file included from /build/amdvlk/src/gpurt/src/shaders/EncodePairedTriangle.hlsl:25:
/build/amdvlk/src/gpurt/src/shaders/EncodePairedTriangleImpl.hlsl:329:38: error: firstbitlow is currently limited to 32-bit width components when targetting SPIR-V
[... 174 lines cut ...]
                     ^
In file included from /build/amdvlk/src/gpurt/src/shaders/BuildBVH.hlsl:47:
In file included from /build/amdvlk/src/gpurt/src/shaders/Common.hlsl:50:
In file included from /build/amdvlk/src/gpurt/src/shaders/Debug.hlsl:30:
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:89: warning: parameter 'timerLo' is uninitialized when used here [-Wparameter-usage]
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
                                                                                        ^~~~~~~~~~~~~~~
/build/amdvlk/src/gpurt/src/shaders/../shared/rayTracingDefs.h:65:29: note: expanded from macro 'DUMMY_VOID_FUNC'
#define DUMMY_VOID_FUNC   { }
                            ^
/build/amdvlk/src/gpurt/src/shaders/Extensions.hlsl:202:64: note: variable 'timerLo' is declared here
__decl void AmdTraceRaySampleGpuTimer(out_param(uint) timerHi, out_param(uint) timerLo) DUMMY_VOID_FUNC
[... cut ...]
Failed to compile Vulkan shader config ShaderConfig< Path: BuildBVH.hlsl, EntryPoint: BuildBVH, OutputName: None, Defines: GPURT_DEVELOPER=1,GPURT_CLIENT_API_VULKAN=1,GPURT_CLIENT_INTERFACE_MAJOR_VERSION=41,USE_BUILD_LBVH=1,GPURT_BVH_BUILD_SHADER=1, isBVH: True >
Launching threads for raytracing shader compilation...
Raytracing shader compilation completed with no errors.
Launching threads for raytracing shader compilation...
```
TL;DR: it seems like whenever it gets to `[1819/2775] Generating pipelines/g_internal_shaders.h, pipelines/g_GpuRtLibrary.h`, it shits the bed.

AnzhongHuang, 2025-02-18:
DXC is used to compile shaders. These errors may occur when an incorrect version of DXC is used. Please try upgrading your DXC version, which can be found in the latest Vulkan SDK.

qiaojbao, 2025-02-18:
/build/amdvlk/src/gpurt/src/shaders/Common.hlsl:570:18: error: firstbithigh is currently limited to 32-bit width components when targetting SPIR-V
    return (63 - firstbithigh(value));
                 ^

Seems you are building with a wrong GPURT source code.

Current 2025Q1.1 here should be
https://github.com/GPUOpen-Drivers/gpurt/commit/c3a4d170f0bacc81df90716887b6546ab41ff5f2
https://github.com/GPUOpen-Drivers/gpurt/blob/dev/src/shaders/Common.hlsl
static int clz64(uint64_t value)
{
    return (63 - FIRSTBITHIGH_U64(value));
}

closed 2025-02-18 by qiaojbao

qiaojbao, 2025-02-18:
Seems you download AMDVLK 2024Q1.1 codes.
The error line be found.
line 570:    return (63 - firstbithigh(value));

HurricanePootis, 2025-02-18:
Ah, my bad. I did a typo, typing a 4 instead of a 5 😄 No wonder it didn't work