GPUOpen-Drivers/AMDVLK #404: Oblivion Remaster - 9070XT

ferrreo, 2025-05-20:
On a 9070XT with latest proton, latest stable kernel, latest GPU firmware etc and any distro Oblivion Remaster causes a GPU hang when using AMDVLK. RADV/mesa is unaffected. Can't seem to find a workaround.

perlfu, 2025-05-21:
Can you provide any additional details to help reproduce the issue, e.g.
- When does the crash occur in game play?
- If this is a GPU hang (screen locks up but game keeps running), can you include dmesg output from amdgpu driver?

ferrreo, 2025-05-21:
> Can you provide any additional details to help reproduce the issue, e.g.
> 
> * When does the crash occur in game play?
> * If this is a GPU hang (screen locks up but game keeps running), can you include dmesg output from amdgpu driver?

No problem

Replication Steps:

1. Load up the game
2. Set everything to ultra
3. Set Hardware Lumen (RT) on - any quality level
4. Load a save or start a new game
5. Watch as frame times take 5000ms+ and gfx corruption occurs

Software lumen is uneffected. I have tested on the latest RC kernel as well as latest stable, some older stable and have tested lastest linux-firmware as well as last released linux-firmware as well as last released firmware from the amd site. IT is the same results for all.

I have attached the amdgpu items in my dmesg log from when it ocurred.

[dmesg_log.txt](https://github.com/user-attachments/files/20369435/dmesg_log.txt)

If there is anything else you need please let me know, happy to help where I can.

Supergamer369, 2025-07-14:
I can confirm this also happens for me on Linux with Hardware Raytracing on.