GPUOpen-Drivers/AMDVLK #413: Vulkan/AMDVLK: Device memory allocation fails when single compute buffer > ~2 GiB (same model works on RADV)

kyuz0, 2025-08-04:
### Problem description & steps to reproduce

There is an open issue on llama.ccp (https://github.com/ggml-org/llama.cpp/issues/15054) for a bug on systems using AMDVLK. Some GGUF models that load and run under RADV fail during model/context initialization in llama.cpp’s Vulkan backend with VK_ERROR_OUT_OF_DEVICE_MEMORY:

```
ggml_vulkan: Device memory allocation of size 2819260416 failed.
ggml_vulkan: Requested buffer size exceeds device memory allocation limit: ErrorOutOfDeviceMemory
alloc_tensor_range: failed to allocate Vulkan0 buffer of size 2819260416
```

The failures correlate with a single Vulkan allocation (the ggml_vulkan compute buffer) exceeding the driver’s per-allocation cap VkPhysicalDeviceLimits::maxMemoryAllocationSize. On the affected machine this is 0x80000000 (≈2 GiB). The backend occasionally needs to request ~2.0–2.6 GiB for this buffer, which succeeds on RADV but is rejected by AMDVLK, indicating the issue is the per-allocation limit, not total VRAM.

The following response from the original llama.cpp issue made me think it was a good idea to raise this here:

> Yes, the driver sets a maximum allocation size and a maximum buffer size limit. This is 2GB on amdvlk and the proprietary AMD drivers, and 4GB on RADV. We have no control over this, I don't know why AMD keeps the limit below the theoretical maximum that all the other major Vulkan drivers use.

###  Example (Gemma 3 27B BF16)

For example, this happens with gemma-3-27b-it-BF16/gemma-3-27b-it-BF16 (https://huggingface.co/unsloth/gemma-3-27b-it-GGUF):

```
$ llama-cli -ngl 99 -fa -m models/gemma-3-27b-it-BF16/gemma-3-27b-it-BF16-00001-of-00002.gguf 
ggml_vulkan: Found 1 Vulkan devices:
ggml_vulkan: 0 = Radeon 8060S Graphics (AMD open-source driver) | uma: 1 | fp16: 1 | bf16: 0 | warp size: 64 | shared memory: 32768 | int dot: 1 | matrix cores: KHR_coopmat
build: 6060 (9c35706b) with cc (GCC) 15.1.1 20250719 (Red Hat 15.1.1-5) for x86_64-redhat-linux
main: llama backend init
main: load the model and apply lora adapter, if any
llama_model_load_from_file_impl: using device Vulkan0 (Radeon 8060S Graphics) - 85720 MiB free
llama_model_loader: additional 1 GGUFs metadata loaded.
llama_model_loader: loaded meta data with 39 key-value pairs and 808 tensors from models/gemma-3-27b-it-BF16/gemma-3-27b-it-BF16-00001-of-00002.gguf (version GGUF V3 (latest))
llama_model_loader: Dumping metadata keys/values. Note: KV overrides do not apply in this output.
llama_model_loader: - kv   0:                       general.architecture str              = gemma3
llama_model_loader: - kv   1:                               general.type str              = model
llama_model_loader: - kv   2:                               general.name str              = Gemma-3-27B-It
llama_model_loader: - kv   3:                           general.finetune str              = it
llama_model_loader: - kv   4:                           general.basename str              = Gemma-3-27B-It
llama_model_loader: - kv   5:                       general.quantized_by str              = Unsloth
llama_model_loader: - kv   6:                         general.size_label str              = 27B
llama_model_loader: - kv   7:                           general.repo_url str              = https://huggingface.co/unsloth
llama_model_loader: - kv   8:                      gemma3.context_length u32              = 131072
llama_model_loader: - kv   9:                    gemma3.embedding_length u32              = 5376
llama_model_loader: - kv  10:                         gemma3.block_count u32              = 62
llama_model_loader: - kv  11:                 gemma3.feed_forward_length u32              = 21504
llama_model_loader: - kv  12:                gemma3.attention.head_count u32              = 32
llama_model_loader: - kv  13:    gemma3.attention.layer_norm_rms_epsilon f32              = 0.000001
llama_model_loader: - kv  14:                gemma3.attention.key_length u32              = 128
llama_model_loader: - kv  15:              gemma3.attention.value_length u32              = 128
llama_model_loader: - kv  16:                          general.file_type u32              = 32
llama_model_loader: - kv  17:                      gemma3.rope.freq_base f32              = 1000000.000000
llama_model_loader: - kv  18:            gemma3.attention.sliding_window u32              = 1024
llama_model_loader: - kv  19:             gemma3.attention.head_count_kv u32              = 16
llama_model_loader: - kv  20:                   gemma3.rope.scaling.type str              = linear
llama_model_loader: - kv  21:                 gemma3.rope.scaling.factor f32              = 8.000000
llama_model_loader: - kv  22:               general.quantization_version u32              = 2
llama_model_loader: - kv  23:                       tokenizer.ggml.model str              = llama
llama_model_loader: - kv  24:                         tokenizer.ggml.pre str              = default
llama_model_loader: - kv  25:                      tokenizer.ggml.tokens arr[str,262208]  = ["<pad>", "<eos>", "<bos>", "<unk>", ...
llama_model_loader: - kv  26:                      tokenizer.ggml.scores arr[f32,262208]  = [-1000.000000, -1000.000000, -1000.00...
llama_model_loader: - kv  27:                  tokenizer.ggml.token_type arr[i32,262208]  = [3, 3, 3, 3, 3, 4, 3, 3, 3, 3, 3, 3, ...
llama_model_loader: - kv  28:                tokenizer.ggml.bos_token_id u32              = 2
llama_model_loader: - kv  29:                tokenizer.ggml.eos_token_id u32              = 106
llama_model_loader: - kv  30:            tokenizer.ggml.unknown_token_id u32              = 3
llama_model_loader: - kv  31:            tokenizer.ggml.padding_token_id u32              = 0
llama_model_loader: - kv  32:               tokenizer.ggml.add_bos_token bool             = true
llama_model_loader: - kv  33:               tokenizer.ggml.add_eos_token bool             = false
llama_model_loader: - kv  34:                    tokenizer.chat_template str              = {{ bos_token }}\n{%- if messages[0]['r...
llama_model_loader: - kv  35:            tokenizer.ggml.add_space_prefix bool             = false
llama_model_loader: - kv  36:                                   split.no u16              = 0
llama_model_loader: - kv  37:                                split.count u16              = 2
llama_model_loader: - kv  38:                        split.tensors.count i32              = 808
[... 243 lines cut ...]
$vulkaninfo | grep -i maxMemoryAllocationSize

'DISPLAY' environment variable not set... skipping surface info
	maxMemoryAllocationSize           = 0xfffffffc
	maxMemoryAllocationSize           = 0x80000000
```

```bash
$vulkaninfo | grep -i maxMemoryAllocationSize

'DISPLAY' environment variable not set... skipping surface info
	maxMemoryAllocationSize           = 0x80000000
```

### Name and Version

llama-cli --version
ggml_vulkan: Found 1 Vulkan devices:
ggml_vulkan: 0 = Radeon 8060S Graphics (AMD open-source driver) | uma: 1 | fp16: 1 | bf16: 0 | warp size: 64 | shared memory: 32768 | int dot: 1 | matrix cores: KHR_coopmat
version: 6060 (9c35706b)
built with cc (GCC) 15.1.1 20250719 (Red Hat 15.1.1-5) for x86_64-redhat-linux

### Operating systems

Linux

lhl, 2025-08-15:
Just an FYI, I was taking a look at this just now and it seems the main issue is that https://github.com/ggml-org/llama.cpp/issues/15120#issuecomment-3190596899

mike-fischer-ml, 2025-10-26:
For the use case of local LLM's , if its not possible allocate enough memory to the vulkan amd device, the AMD gpu/apu's really can't run any decent local model. 

If you read the llama cpp github issues, theres a decent of amount problems related to memory allocations. 

Might want to spend some time on this.